JetBrains/intellij-micropython #91: Unable to add MicroPython facet in IntelliJ IDEA 2019.1.3

pbackx, 2019-07-11:
For some reason, the MicroPython facet is not showing up in my project. Does the project need to confirm to a certain file structure? Or are there other requirements?

[image]

IDEA details:
IntelliJ IDEA 2019.1.3 (Ultimate Edition)
Build #IU-191.7479.19, built on May 28, 2019
JRE: 1.8.0_202-release-1483-b58 amd64
JVM: OpenJDK 64-Bit Server VM by JetBrains s.r.o
Windows 10 10.0

PRosenb, 2020-01-16:
I have the same issue under OSX. Did you find a solution @pbackx or is this project abandoned?

pbackx, 2020-01-17:
I haven't really retried it since reporting the issue, so I have no idea. I do see there are some commit's regarding the new IDEA version, so they may not have been released? But, no, sorry, can't help I'm afraid.

vlasovskikh, 2020-06-05:
@pbackx @PRosenb Sorry for my late response. The current version of the intelllij-micropython plugin no longer supports IntelliJ 2019.1. I've just checked it on IntelliJ 2020.1 and had no issues with adding the MicroPython facet. If you still have these problems, feel free to comment.

closed 2020-06-05 by vlasovskikh