JetBrains/intellij-micropython #92: Flash in PyCharm working but running on ESP8266

tuzzo77, 2019-07-30:
I'm flashing simple code to my ESP8266. Flash seems to be all correct:

Connecting to COM3
Uploading files: 0% (0/1)
C:\Users\alex\PycharmProjects\ESP8266\Test.py -> Test.py
Uploading files: 100% (1/1)
Soft reboot

Bu code not running on my board.
Same code using ESPlorer IDE working,

Thanks

ivkond, 2019-08-10:
@tuzzo77, use following name conventions for your files: 

- boot.py – this script is executed when the pyboard boots up.
- main.py – this is the main script that will contain your Python program.

Source: [Running your first script](http://docs.micropython.org/en/v1.9.3/pyboard/pyboard/tutorial/script.html)

neuberfran, 2019-08-29:
https://forum.micropython.org/viewtopic.php?f=18&t=6857&p=38966#p38966****

vlasovskikh, 2020-06-05:
@tuzzo77 I hope the replies above helped you with your question. Thanks @ivkond and @neuberfran for your comments!

closed 2020-06-05 by vlasovskikh