JetBrains/intellij-micropython #94: Are there any plans for ESP32 support?

mawg, 2019-09-16:
I think that ESP32 must be pretty popular by now. Are there any plans for support?

oinsd, 2019-09-20:
I want to ask that too.

ivkond, 2019-09-27:
https://github.com/vlasovskikh/intellij-micropython/pull/88

MathyV, 2019-11-16:
It just works if you select the ESP8266 device type, which makes sense as basically they're both running MicroPython. The referenced pull request doesn't actually add anything for ESP32 support as far as I can tell, it just replaces the usage of adafruit-ampy.

The only thing that could be added is probably better code hints for ESP32 but since I just started using the ESP32 myself I have no idea if they differ that much from ESP8266.

vlasovskikh, 2020-06-05:
Yes, there are, follow #44.

closed 2020-06-05 by vlasovskikh