JetBrains/intellij-micropython #95: PyCharm can't see my BBC micro:bit

mawg, 2019-09-17:
My problem is that after installing the micropython plugin, it won't recognize my BBC micro-bit "Detect" does not work, and when I brows to the device's WIndwos folder, the "OK" button is not enabled.

[image]

[image]

[image]

vlasovskikh, 2020-06-05:
@mawg Sorry for my late reply. Is this issue still relevant to you? I'm sorry for confusion, but you have to specify the COM port which was assigned to your device by Windows. It's COM4, COM5, or COM + some other number. At the moment I don't have an access to a Windows machine so I cannot tell you why "Detect" button failed to detect it or how you can do it by yourself. I could offer only a naive piece of advice to google how to determine a COM port number on Windows for your BBC Micro:bit.

closed 2020-06-05 by vlasovskikh

msuleman67h, 2020-10-17:
Hey, I did put the right COM in the path but it gave me this error

> Configuration is still incorrect. Do you want to edit it again?

If I continue anyways, nothing happens. I used [MicroBitTools](https://github.com/SWZ-github/MicroBitTools). It flashed without any issue.