JetBrains/intellij-micropython #99: ampy.pyboard.PyboardError: failed to access COM5

caole08, 2020-01-13:
C:\Users\79966\Anaconda3\python.exe C:\Users\79966\.PyCharm2019.1\config\plugins\intellij-micropython/scripts/microupload.py -C C:/Users/79966/PycharmProjects/untitled2 -v COM5 C:/Users/79966/PycharmProjects/untitled2/main.py
Connecting to COM5
Traceback (most recent call last):
  File "C:\Users\79966\.PyCharm2019.1\config\plugins\intellij-micropython/scripts/microupload.py", line 138, in <module>
    main(sys.argv[1:])
  File "C:\Users\79966\.PyCharm2019.1\config\plugins\intellij-micropython/scripts/microupload.py", line 56, in main
    board = Pyboard(port)
  File "C:\Users\79966\Anaconda3\lib\site-packages\ampy\pyboard.py", line 147, in __init__
    raise PyboardError('failed to access ' + device)
ampy.pyboard.PyboardError: failed to access COM5

patfrench, 2020-03-25:
Hi,
I have a same problem, same error. with anaconda3

[image]

[image]

FLASH : problem
REPL : ok

thank you

vlasovskikh, 2020-06-05:
Sorry for my late reply. Does the "Detect" button work in the MicroPython settings next to the "Device path" on your screenshot? Does it determine the same COM device path that you specify manually? I suppose that your Windows assigned a different COM number to your device. Sorry I don't have a Windows machine available at this moment, so I cannot reproduce this issue.

vlasovskikh, 2020-06-13:
Closing this issue as obsolete. Please feel free to comment it if the problem persists.

closed 2020-06-13 by vlasovskikh