JetBrains/intellij-micropython #101: Could you add termainal quick access by using sidebar?

CHN-STUDENT, 2020-04-14:
Just like this.
Thanks!
[image]

CHN-STUDENT, 2020-04-14:
By the way, i do not know why my keyborad backspace key can not use on repl? Could you tell me? Thanks!

vlasovskikh, 2020-06-05:
@CHN-STUDENT Thanks for your suggestion to add the MicroPython REPL tool window. At the moment the "MicroPython REPL" action actually opens another tool window: "Terminal", but I like the idea of embedding the terminal in a separate "MicroPython REPL" tool window to make it more accessible.

ChrisH861, 2020-06-15:
Would also be nice if any open REPL terminals automatically closed when you flash a file and then reopened afterwards.

vlasovskikh, 2020-06-15:
@ChrisH861 Please follow #53 for that.

vlasovskikh, 2020-09-21:
A simplistic code snippet for adding the terminal in a tool window:
```
  private static void addTerminal(@NotNull ToolWindow toolWindow,
                                  @NotNull AbstractTerminalRunner<?> terminalRunner,
                                  @NotNull @Nls String tabName) {
    Content content = ContentFactory.SERVICE.getInstance().createContent(null, tabName, false);
    JBTerminalWidget terminalWidget = terminalRunner.createTerminalWidget(content, null);
    content.setComponent(terminalWidget.getComponent());
    content.setPreferredFocusableComponent(terminalWidget.getPreferredFocusableComponent());
    toolWindow.getContentManager().addContent(content);
  }
```

lensvol, 2021-01-30:
Can I take this over? Seriously, this annoys me to no end.

vlasovskikh, 2021-01-30:
@lensvol I don't have plans for the near future to implement it, feel free to take over.

lensvol, 2021-01-31:
Can you please assign it to me, then? :)