JetBrains/intellij-micropython #102: Typehints missing classes (ADC, PWM, and probably more)

1oglop1, 2020-04-26:
Hi,

I recently started playing with ESP32 and MicroPython. 
This plugins looks like a great idea to make things easier however there are some missing modules for ESP8266 and ESP32.

So I did some reading on how this plugin works and found all type hints to be updated 2 years ago, which made me wonder whether they are maintained manually or generated from MicroPython code.

I came across this tool https://github.com/Josverl/micropython-stubber which helps to create pyi files directly on the device itself.

I'm still not sure how to use the stubber, however my idea is to use it to generate stubs and contribute them to this project or to `typeshed` to keep the type hinting up to date with less effort. 

@vlasovskikh 
What do you think about this idea?
Would you be willing to give me a hand to make this happen?

vlasovskikh, 2020-06-07:
@1oglop1 Thanks for trying out the plugin! I've created the stubs in intellij-micropython manually. I was not aware of micropython-stubber, thanks for sharing the link and for suggesting an idea to use it. I believe it will be helpful to get stubs for missing modules or to generate updates for existing stubs. There are however one thing that is better with manually created stubs: I put docstrings into my stubs based on the published docs on MicroPython modules. I find having docs on mouse hover / <kbd>Ctrl+Q</kbd> / <kbd>F1</kbd> very helpful.

I like the idea of committing stubs generated by micropython-stubber to intellij-micropython. It might also be possible to integrate micropython-stubber so that intellij-micropython invokes it for introspecting new modules for which there are no stubs. I'm open to discussing other ideas on improving the situation with stubs as well.

1oglop1, 2020-06-09:
Yes the docs via quick docs is one of my motivations to do this as well.

I will try the stubber with my ESP32 and let you know as soon as I get some results.

w0nd4bra, 2020-06-16:
Hey,
came across this discussion, tried the mentioned tool in the first post and get the files from my ESP32 with ampy... just tested it with one module and it worked for me, but I dont know if its right, cause its beyond my knowledge. 

Maybe its easier for your work, but as I said before, Ive never did such stubber thing before, I was just interested :D

greetings

1oglop1, 2020-06-17:
@w0nd4bra I haven't use the stubber before as well, if you have managed to get an output it would worth sharing! 
If you are interested in live chat you could either join our PyAmsterdam slack (https://py.amsterdam) or we could use gitter.

I may have some time this weekend to play around.

w0nd4bra, 2020-06-17:
I can upload the files here on Github if you want. There is nothing special to talk about it,i just want to make you guys to make the plugin better, i cant do it. Just want to use the plugin and pycharm to have some fun with my esp32.

Have some ESP-01 laying around, will check later if its the 1mb version or just the 512k one (so i can flash MicroPython). So we have some files for the ESP8266 too.

vlasovskikh, 2020-06-18:
@1oglop1 @w0nd4bra I've created [a Gitter community chat](https://gitter.im/intellij-micropython/community) about the plugin, feel free to join and discuss things.

tobiasfunke1, 2020-09-27:
Maybe this could help: https://pypi.org/project/micropy-cli/ ?

thoastbrot, 2020-09-30:
I went the road for the least effort for some value and ended up declaring some stubs here:

https://github.com/thoastbrot/intellij-micropython/blob/master/typehints/micropython/machine.pyi

I did not add documentation yet, as I didn't look up whether the actual API has some or anything... but opening the file in intellij, applying autoformat for the sake of it and copy-pasting APIs from micropython docs and the open python shell... it was worth it. For the lack of docstrings, I did not open a pull request yet. If you feel like adding those, please do so (and open said pull request - I'd be happy to support with code issues).

gpsgarage-ru, 2021-01-24:
joining in to request. Without ALL typehints this plugin is more a mock on the users than something useful.

hlovatt, 2021-08-08:
I'm creating high quality typesheds/stubs including documentation as well as types, see https://github.com/hlovatt/PyBoardTypeshed and then hopefully they will all be included in the Micropython plugin (many of the existing typesheds/stubs in Micropython come from this project). Give me a few weeks and I should have done the complete stack for a PyBoard at least :).