JetBrains/intellij-micropython #103: Auto Detect Broken in V1.1

jdjjm, 2020-06-07:
Just updated to version 1.1 today and found the auto detect to be broken for a ftdi programmer.

On the old version the Detect button would find my ftdi programmer but this time it does not.
Plugging in a ESP8266 directly works fine so it seems to be just an issue with the fdti. 

Would love to see this fixed as I use an fdti programmer for custom ESP boards and auto detect is much easier than specifying the port each time it changes.

Thanks

Micropython Plugin: V1.1
Windows 10
Pycharm 2020.1.1

vlasovskikh, 2020-06-07:
@jdjjm Thanks for your report!

How do you feel about sending a one-line PR to the plugin? Alternatively I can just do it by myself.

If you're interested, the place where the USB IDs for ESP8266 are listed is [here](https://github.com/vlasovskikh/intellij-micropython/blob/6fee7d5df31d3c2933354eb7ab5dd21a49f26ad6/src/main/kotlin/com/jetbrains/micropython/devices/Esp8266DeviceProvider.kt#L27) in the code base. I have to ask you to electronically sign a contributor license agreement at http://www.jetbrains.com/agreements/cla/ Also it would be great if you built the plugin from sources as described in the [contribution docs](https://github.com/vlasovskikh/intellij-micropython#source-code) and checked it with your device.

jdjjm, 2020-06-08:
I made the changes and built the plugin and it works. I have added a pull request like you asked.

vlasovskikh, 2020-06-08:
Fixed by @jdjjm. Thanks!

closed 2020-06-08 by vlasovskikh