JetBrains/intellij-micropython #109: how star code without flashing?

BigStupidBeast, 2020-06-21:
if i use Thonny IDE(or Mu or ESPlorer) i push button F5 and code from current file starts without
flashing. how to do the same trick in pyCharm?

vlasovskikh, 2020-06-21:
@BigStupidBeast Do you mean it runs on your computer or on your MicroPython device?

BigStupidBeast, 2020-06-22:
Hi, Andrey!
I mean it runs on my ESP32. Because i can use in my code files that were flashed on my device earlier. not even speaking about timers, led blinking, pin interrupts and datas from sensors...
I try to start PyCharm+Micropython on Windows7 and on Mac(catalina). I failed both times. Many times:-).
I realy want to use PyCharm, but right now i setup up Thonny, open it, select type of micropython device and start coding. Simple, fast and without shamanistic dance . 
p.s. The Mu does not always detect the device. The ESPlorer sometimes flash files with errors.

vlasovskikh, 2020-06-22:
@BigStupidBeast When you run "Flash foo.py" via the context menu of your file named foo.py, the MicroPython plugin actually flashes (or should I say "uploads") only foo.py, without chaining or deleting any other files on your device. It also respects paths relative to your project root or any directories you've marked as source roots. So it seems that for your project it will be enough to just run the "Flash ..." configurations only for the files you're interested in uploading to your ESP32 device. Please make sure that you've selected "ESP8266" as your MicroPython device in your Settings | MicroPython. Also make sure you're using the MicroPython plugin 1.1, the most recent release.

Does it answer your question?

closed 2020-09-20 by vlasovskikh