JetBrains/intellij-micropython #110: Micropython language does not appear in Intellij

arpadtamasi, 2020-06-21:
Hi, 

I've installed the plugin and trying to use it but MicroPython does not appear in the language list and is disabled under tools.
How can I activate it?

[image]

vlasovskikh, 2020-06-21:
@arpadtamasi Is the MicroPython plugin enabled in your Settings | Plugins?

arpadtamasi, 2020-06-22:
Sure

—

Arpad Tamasi
Principal Product manager at Rollbar

vlasovskikh, 2020-06-22:
I’ve missed that you’re using IntelliJ, not PyCharm. The steps for setting up MicroPython support are different for them described in the docs:

> IntelliJ: Add the MicroPython facet to a Python module in your project structure and specify the path to your MicroPython device

So it’a in the Project Structure, not in Settings. See README.md for more details.

closed 2020-06-22 by vlasovskikh

arpadtamasi, 2020-06-22:
Thanks a lot, it works.