JetBrains/intellij-micropython #111: Error message when Run 'Flash main.py' entered after changes made

Scissorunner, 2020-06-28:
Hello,
Hope I didn't miss this as a previously corrected issue. I've been following this repository for some time and haven't seen this addressed.

I have a simple blink program I'm using for testing. Everything works as it should when file is initially flashed to ESP8266 device. I check file operation with the Tools/MicroPython/MicroPython REPL, terminal screen and all is well.

If I go back into my file and make a change, such as sleep() time and then use flash to send to my device I get the following error. I've tried Ctrl-C to first stop the running program but cannot get past the error.

If I File/Close Project and then reopen the project I can successfully flash the file and everything works.

Why can't I make a change and flash without resorting to closing the project and reopening the project? I'm guessing something to do with defaulting to this 'board = Pyboard(port)'

Connecting to COM4
Traceback (most recent call last):
  File "C:\Users\Craig's PC\AppData\Roaming\JetBrains\PyCharmCE2020.1\plugins\intellij-micropython/scripts/microupload.py", line 139, in <module>
    main(sys.argv[1:])
  File "C:\Users\Craig's PC\AppData\Roaming\JetBrains\PyCharmCE2020.1\plugins\intellij-micropython/scripts/microupload.py", line 56, in main
    board = Pyboard(port)
  File "C:\Users\Craig's PC\PycharmProjects\ESP Projects\venv\lib\site-packages\ampy\pyboard.py", line 147, in __init__
    raise PyboardError('failed to access ' + device)
ampy.pyboard.PyboardError: failed to access COM4

vlasovskikh, 2020-06-30:
@Scissorunner Do you have your REPL or any other connections to your device open while you're flashing your main.py for the second time?

Scissorunner, 2020-06-30:
@vlasovskikh Yes, previously tested I had the REPL open but had done a Ctrl-C to stop the program.

New testing with REPL closed (Terminated) I have no problem making changes and flashing main.py multiple times. I can reopen REPL to check/debug code and after checking close REPL, change code and flash main.py with zero errors.

Problem solved! Thank you so much for your work on this... GREATLY APPRECIATED!

closed 2020-06-30 by Scissorunner

vlasovskikh, 2020-07-01:
Follow #53 for closing the REPL session on flashing files to the device.