JetBrains/intellij-micropython #112: not able to import utime

jasonz2k, 2020-06-30:
In PyCharm CE 2020.1, utime could not be imported in the python code. The micropython is set for microbit.

w0nd4bra, 2020-07-10:
Because there are missing stubfiles for the typehints.
See #102 for that. It's just some annoying work to do, because lot of copy and paste and repetitive work. Nothing to enjoy, but some guys (like me) working on it. I just finished with the whole Micropython libs.
It just take some time.

hlovatt, 2021-08-08:
it's on my list to create for https://github.com/hlovatt/PyBoardTypeshed and then hopefully it will be included in the Micropython plugin (many of the existing typesheds/stubs in Micropython come from this project). Give me a few weeks :).

vlasovskikh, 2021-12-01:
Should be fixed by #176.

closed 2021-12-01 by vlasovskikh