JetBrains/intellij-micropython #113: Pressing backspace in REPL is generating <-[K

Architk08, 2020-07-09:
I've recently installed pycharm and the micropython plugin is v1.1.1 on windows 7
The backspace key is echoing back "<-[K" character.  Also, multiple up arrows echo some random characters as well.
Is there any fix for this issue?
[image]

CaseyKnauss, 2021-03-05:
Same behavior for me on Windows 10

DannyOnkies, 2021-03-15:
Hi .... the same thing happens to me too 
[image]

sandyscott, 2021-08-09:
So on my system each press of the backspace key removes the previous character, then adds these 3 characters:`←[K`. Subsequent presses remove the `K`, then add the same 3 again, so you end with a sequence of the arrow-squarebracket-arrow-squarebracket that ends with a `K`.

The microcontoller does appear to be correctly dealing with the keypresses, the issue is with the display in PyCharm.
ie.
1. Type `print("abc_def`
2. Press backspace 3 times
3. Type `xyz")` then press enter.
This is equivalent to entering `print("abc_xyz")`
4. The result is:
```
>>> print("abc_de←[←[←[Kxyz")
abc_xyz
```

5. `abc_xyz` is printed as expected.

PyCharm 2021.2, MicroPython Plugin 1.2, Raspberry Pi Pico running MicroPython v1.16