JetBrains/intellij-micropython #117: Linux REPL from tool menu not working

palmtreefrb, 2020-08-02:
In the latest releases microrepl.py is not executing from tool menu. It just opens an empty terminal.
I can manually execute from terminal command line...

frb@ubuntu:~/.local/share/JetBrains/PyCharm2020.2/intellij-micropython/scripts$ python3 microrepl.py /dev/ttyUSB1
Device path /dev/ttyUSB1
Quit: Ctrl+] | Stop program: Ctrl+C | Reset: Ctrl+D
Type 'help()' (without the quotes) then press ENTER.

>>>

cr0mbly, 2020-08-02:
This is happening to me as well.

gbaranski, 2020-08-03:
Same issue on MacOS Catalina

hlovatt, 2020-08-10:
Similar issue MacOS Catalina, PyCharm CE 2020.2, & plugin 1.1.1.

When attempt to open MicroPython REPL I get:

```
L8 n
/Applications/PyCharm CE.app/Contents/plugins/terminal/.zshrc:source:45: no such file or directory: /Users/lov080/Google
```

Is this because `Google Drive` has a space in it? 

Also note strange `L8 n`?

Also:

```
% python3 microrepl.py /dev/tty.usbmodem2067346A344D2 
Traceback (most recent call last):
  File "microrepl.py", line 32, in <module>
    from serial import Serial, SerialException
ModuleNotFoundError: No module named 'serial'
```

vlasovskikh, 2020-09-20:
@palmtreefrb Thanks for your bug report! I've found a regression in the terminal subsystem of the IntelliJ platform and came up with a workaround. I'll publish the fix soon.

closed 2020-09-20 by vlasovskikh

palmtreefrb, 2020-09-21:
Ok, looks good!!! Thank you...
[image]

vlasovskikh, 2020-10-24:
I've released 1.1.2 with the fix.