JetBrains/intellij-micropython #119: No REPL access on Windows

HexVitor, 2020-08-16:
_Environment:_
```
Board: ESP32 (ESP-WROOM-32, chip ESP32D0WDQ6, revision 1)
Firmware: MicroPython v1.11 (2019-10-26)
IDE: PyCharm Community 2020.2
Python: 3.8.3 (x64)
OS: Windows 10 Home Single Language 2004 (x64)
```

_Tools > MicroPython > MicroPython REPL_ returns 2 local terminals (Local and Local(2)) and both display:
```
Microsoft Windows [versão 10.0.19041.450]
(c) 2020 Microsoft Corporation. Todos os direitos reservados.

(venv) C:\Users\vitor\PycharmProjects\ESP32>
```

The board is recognized in COM5 and I can flash/run codes by receiving the following messages:

```
C:\Users\vitor\PycharmProjects\ESP32\venv\Scripts\python.exe C:\Users\vitor\AppData\Roaming\JetBrains\PyCharmCE2020.2\plugins\intellij-micropython/scripts/microupload.py -C C:/Users/vitor/PycharmProjects/ESP32 -v COM5 C:/Users/vitor/PycharmProjects/ESP32/main.py
Connecting to COM5
Uploading files: 0% (0/1)
C:\Users\vitor\PycharmProjects\ESP32\main.py -> main.py
Uploading files: 100% (1/1)
Soft reboot
```

I am very optimistic about PyCharm for MicroPython, but unfortunately I am having this error that until then I have not been able to solve.
I updated the files findusb.py, microrepl.py, microupload.py and miniterm.py, installed all the requested requirements (adafruit-ampy, esptool, pyserial...), enabled compatibility with python 3.5 and even so I was not successful.
I will be happy if someone can help me and even more if I can solve this problem.
Thank you for your attention.

jdjjm, 2020-08-17:
This plugin has not been updated for PyCharm 2020.2 yet. It seems the 2020.2 release has broken the REPL access. You can roll back to 2020.1 until the plugin has been updated for 2020.2 if you need REPL access.

vlasovskikh, 2020-09-20:
@HexVitor @jdjjm Thanks for reporting it! It's reproducible on all the platforms. I've marked it as a duplicate for #117.

closed 2020-09-20 by vlasovskikh