JetBrains/intellij-micropython #120: Unable to run any micropython in PyCharm for ESP8266

AstroDrabb, 2020-09-30:
I was able to install the Micropython plugin into PyCharm 2020.2, however I can't get even the simplest of examples to work.  Example code:

```
import machine

def test(name) -> None:
    # Use a breakpoint in the code line below to debug your script.
    print(machine.freq())  # Press Ctrl+F8 to toggle the breakpoint.
    machine.freq(160000000)
    print(machine.freq())

if __name__ == '__main__':
    test()
```
```
Traceback (most recent call last):
  File "/home/jim/code/PycharmProjects/micropy-d1mini/main.py", line 6, in <module>
    import machine
ModuleNotFoundError: No module named 'machine'

Process finished with exit code 1
```
What step(s) am I missing to test and upload script to run?

MyDataIsReady, 2020-10-04:
Had the same problem. Is your board being detected? Try to uninstall and install the plugin. Double-check dependencies are being installed without issues.

AstroDrabb, 2020-10-04:
@MyDataIsReady I am using Linux, so no issues with drivers that I am aware of.  I will try to uninstall the plugin and reinstall.

jdjjm, 2020-10-07:
@AstroDrabb by the look of your error message it looks like you are trying to run micropython code with regular python. You must run it on a micropython enabled microcontroller. Make sure you have enabled micropython support for the project by doing `Settings > Languages & Frameworks > MicroPython > Enable MicroPython Support`
If you right click on the file you have created you will now have the option to `Run 'Flash main.py'`. This will flash your script to your connected micropython device so it can run on there, not on your computer.

vlasovskikh, 2020-10-24:
I agree with @jdjjm about the most likely source of this problem. Thanks @jdjjm for answering!

Closing this issue as answered.

closed 2020-10-24 by vlasovskikh

ram19890, 2020-12-17:
Please reopen this issue to investigate, the plugin failed to identify the actual board esp8266(It doesn't show "Flash main.py", it only shows Run 'main', using the default pyhon3.8). As you have mentioned already on the wiki that this board isn't supported, if in that case you can ignore my request! I am on MacOS 10.14.6 with pycharm 203.5981.165 installed, current mircopython plugin installed.

hcet14, 2021-07-01:
I have the same problems. I'm using ESP8266 and ESP32. See also [https://forum.micropython.org/viewtopic.php?f=15&t=10771](url)
[image]
[image]
[image]
Does the plugin tries to load the wrong board?