JetBrains/intellij-micropython #121: MicroPython REPL Not Working in Pycharm - takes me to virtual env.

states-lee-d, 2020-10-05:
Hi,

Each time I try and run a serial connection with my board in Pycharm using the Micropython REPL tool, Pycharm takes me inside '(venv) C:\Users\xxxx\PycharmProjects\setup>'. I've followed instructions from multiple tutorials, yet not much seems to work. I'm on Windows 10.

Thanks

bachi76, 2020-10-06:
Same here, Ubuntu.

TishchenkoArtem, 2020-10-12:
Same problem, Windows 10

functionpointer, 2020-10-16:
Same here, Windows 10. I am using the system interpreter and it just gives me a windows cmd shell in the project folder.

Akshaya227, 2020-10-19:
@states-lee-d  I had the same issue. It was due to the version of Pycharm. The feature is not yet supported in 2020.2 it seems, so rollback to 2020.1 is suggested

mathiaspez, 2020-10-21:
Same here, Pycharm 2020.2.3 under windows 10.

fdesoye, 2020-10-23:
I currently work around with a manual REPL doing `screen /dev/tty.wchusbserial1420  115200` on a Mac with PyCharm 2020.2.3.
But would be great if I wouldn't have to apply this workaround.

vlasovskikh, 2020-10-24:
The fix will be available soon in v1.1.2. Closing it as a duplicate of #117.

closed 2020-10-24 by vlasovskikh

vlasovskikh, 2020-10-24:
I've released 1.1.2 with the fix.

mathiaspez, 2020-10-24:
Thank you!!!
Works like a (py)Charm! ;-)
Thank you again!