JetBrains/intellij-micropython #123: Unable to run any micropython in PyCharm for ESP32

Quadrangle, 2020-10-23:
I installed Python 3.9, PyCharm 2020.2.3CE, the micropython plug in.
My ESP32 board enumerates as COM4 on my Windows10 PC.
My project is setup for micropython, though I find it disturbing that there is no ESP32 choice, so I use ESP8266 as others seem to do for the ESP32 which is in the same family hardware-wise.  But can someone explain why that is not a bug?

I have studied posts here and saw virtually the same problems happened about four months and was closed as not repeatable.  My problem is 100% repeatable.  I noticed there was some discussion about maybe some other window was open and that stopped flashing but I have checked that thoroughly.  I have this same problem on two boards.

When I try to download, I always get:
Connecting to COM4
Uploading files: 0% (0/1)
E:\ ... \micro\main.py -> main.py
b'\x1b[0;32mI (2153) phy: phy_version: 329, Feb 22 2017, 15:58:07, 0, 0\x1b[0m\r\nI (2154) wifi: mode : softAP (10:52:1c:66:87:99)\r\nI (2157) wifi: mode : sta (10:52:1c:66:87:98) + softAP (10:52:1c:66:87:99)\r\nI (2160) wifi: mode : softAP (10:52:1c:66:87:99)\r\n'
Traceback (most recent call last):
  File "C:\Users\wrewi\AppData\Roaming\JetBrains\PyCharmCE2020.2\plugins\intellij-micropython/scripts/microupload.py", line 139, in <module>
    main(sys.argv[1:])
  File "C:\Users\wrewi\AppData\Roaming\JetBrains\PyCharmCE2020.2\plugins\intellij-micropython/scripts/microupload.py", line 79, in main
    files.put(remote_path, fd.read())
  File "E:\Data\Hobby\MotionAndLight\molifirmware\micro\venv\lib\site-packages\ampy\files.py", line 208, in put
    self._pyboard.enter_raw_repl()
  File "E:\Data\Hobby\MotionAndLight\molifirmware\micro\venv\lib\site-packages\ampy\pyboard.py", line 192, in enter_raw_repl
    raise PyboardError('could not enter raw repl')
ampy.pyboard.PyboardError: could not enter raw repl

jdjjm, 2020-11-02:
Make sure you have closed the REPL (Seems like you already know this but just make sure). Have you ever been able to upload code to the board or is this just after a fresh flashing of micropython?

Quadrangle, 2020-11-02:
Hi Andrey,Thank you for your message.  I really admire IntelliJ and JetBrains, and even more so given a person of your stature should reach out to me for feedback.
I do not want to waste your time, so I am reinstalling PyCharm with the Micropython plug-in, and will check a few things before providing what I hope is accurate information.   I promise not to abuse this opportunity by hassling you with emails.
Thanks,Bill Ewing

Quadrangle, 2020-11-02:
Hi Andrey,My feedback below.Best regards,Bill Ewing
1. If there was just one bit of feedback I could convey, it would be: I wish PyCharm worked like uPyLoader.  Although it is an exceedingly simple app, uPyLoader makes everything transparent.  It is very clear what files are in the device and transfers them in either direction with no drama.  It just works.  It will never have an editor or refactoring like PyCharm, but what it does, it does well.  Also, their FAQ is exemplary because it goes into problems that are not their fault.  Please take a look at: 
https://forum.micropython.org/viewtopic.php?t=2245

2. After Run, Run... , the Stop menu is disabled.  The way I can find to stop is to close the Local window, getting a "do you want to terminate..." message.  This seems strange, am I missing something?  Does the Stop menu/button ever work?

3. After Run, Run...  we get messages about connecting to COMx, ... , Soft reboot.  This is great, but other products put us into a useful (REPL) window where we can see things printed, etc.  Is this actually missing from PyCharm?  May I suggest in Run Configuration that there be checkbox options to put us into REPL? 

4. Often, after clicking on Tools, Micropython, Micropython REPL, we get messages like:
   Device path COM4
   Found the device, but could not connect via port 'COM4': could not open port 'COM4': 
       PermissionError(13, 'Access is denied.', None, 5)
   I'm not sure what to suggest. :-(

Workarounds for this are awkward and strange, and erode my team's confidence in PyCharm.  Couldn't PyCharm figure out what the user wants and do something smart?  We do not have any other programs using the COM port.  We told PyCharm what port to use and then it effectively crashes when we click on something that is not disabled.  This is an eggregious design flaw.

It is a sad day for software developers when we tell users to disconnect/reconnect USB.  And yet, we now use switches that open/close +5V in the USB cable that connects an ESP32 to my PC, and this helps.

5. In Settings, Languages & Frameworks, MicroPython, Device type... we would like to suggest 'ESP8266' be changed to include 'ESP32', or perhaps better, make 'ESP32' a new device type (even if it is, at least for now, the same as the 8266 as far as PyCharm is concerned).

6. In the Edit Configurations dialog, the Fix button appears and invokes installation of needed packages.  Very nice feature!  I would like to suggest that the Fix button disappear after fixes are successfully installed.

7. Before I tried PyCharm and Micropython, I viewed the only tutorial available.  It warned me that the "Auto-detect device path" feature would not work.  Months later, it still does not work.  I think the person who wrote the tutorial may be a JetBrains employee.  It is easy enough to figure out what COM port to use, but this is another point of confidence erosion.  Maybe you should remove the automatic stuff.  As a Java developer I would like to tell you how much happier my team is after we changed from the loathsome RxTx to jSerialComm.  Our handling of serial comm problems went from dumb to pretty good.
https://fazecast.github.io/jSerialComm/

    On Monday, November 2, 2020, 11:15:33 AM PST, Bill Ewing <wrewing001@yahoo.com> wrote:  
 
  Hi Andrey,Thank you for your message.  I really admire IntelliJ and JetBrains, and even more so given a person of your stature should reach out to me for feedback.
I do not want to waste your time, so I am reinstalling PyCharm with the Micropython plug-in, and will check a few things before providing what I hope is accurate information.   I promise not to abuse this opportunity by hassling you with emails.
Thanks,Bill Ewing

jdjjm, 2020-11-03:
Keep in mind this is not an Officially Supported Feature of PyCharm but rather just a JetBrains employee that was nice enough to create the micropython plugin. Also I am not @vlasovskikh, I am just another user of the plugin. If you look at the issue in browser instead of through your email, it may be easier for you to see everything and reply (Email formatting can be a bit messy).
Here is a link : https://github.com/vlasovskikh/intellij-micropython/issues/123

That being said, If there is a specific feature that you would like to see added, submit a new issue with the idea and it can be discussed and @vlasovskikh can decide if it should be implemented. Or even better... implement it yourself and submit a PR. Like I said earlier this is not an offical PyCharm product so no one is being paid to work on this, which is why community development is essential if you want to see the plugin grow.

Regarding 2. I do not fully understand what you mean but keep in mind when you Click Run on a MicroPython Configuration it calls a script to upload your code to a micropython enabled device. This means the stop button can only be clicked when the code is being uploaded. Once the code is uploaded the stop button cannot be pressed because the script has terminated after it did its job. In your case the ESP32 is now running the code, not PyCharm, after uploading (hence no stop button).

Regarding 3. see #83 

In regards to 4. the access denied error usually happens when something else is already connected on a specific com port. For example... you already have another REPL window open, or a different serial connection, or code is being uploaded etc. Also make sure you are not using uPyLoader at the same time. Doing that will also give the error.

For point 5 see #44 

Regarding 7. The auto detect does work on certain boards. I use it personally (On windows) and it works great. It will depend on the USB Chip on the device whether or not the script can find it. But there is a checkbox for auto path for a reason. If its not working for you just don't use auto detect and just set one manually. Also (Assuming your using windows) go to Device Manager then choose the Com port driver that appears when the device is connected. In this dialog go to Details, then change the property to Hardware IDs. This should give you a value with a VID number and a PID number. If you have different values to any of the following pairs, then you can let me know and I can create a PR to add your device to the list of ones it scans for.
0x1A86, 0x7523
0x10C4, 0xEA60
0x0403, 0x6001

Also for future reference probably don't give feedback in this format. If you have feedback, create issues on GitHub about specific things you want to see implemented. It just makes life easier for everyone if things are structured and each issue is about one topic only 😄. Also before you post ideas make sure a similar GitHub issue doesn't already exist because there are quite a few ideas already.

Regarding your original issue, you never supplied any information regarding the questions I asked. Is this still an issue? Can you upload any code to the device? Have you tried erasing the flash of the ESP32 and re-flashing micropython with esptool and then trying to upload again?

closed 2020-11-26 by Quadrangle