JetBrains/intellij-micropython #127: Latest micro:bit hex files

microbit-carlos, 2020-12-04:
Hi @vlasovskikh,

There is a new version of micro:bit and as it uses a different microcontroller (nRF52833), so there is a new port of MicroPython as well.

The source code for the V2 port lives in this repository, but our plan is to push it to the upstream MicroPython repository when it's ready, so this one is just temporary: https://github.com/microbit-foundation/micropython-microbit-v2/

The latest release is v2.0.0-beta.2,  but there might be new updates in the near future. While we still use this repository all releases will be published as GH releases:
- https://github.com/microbit-foundation/micropython-microbit-v2/releases/tag/v2.0.0-beta.2
- https://github.com/microbit-foundation/micropython-microbit-v2/releases/

To easily support users and editors we've created a new format of hex file called Universal Hex. This format can pack the hex data for V1 and for V2, so a single hex file can be flashed into any version of the micro:bit.

More info about the Universal Hex format can be found in the micro:bit Tech Site: https://tech.microbit.org/software/hex-format/

There is no "official" Universal Hex release for MicroPython, but a Universal Hex can be easily created with this tool: https://tech.microbit.org/software/universal-hex-creator/

To go with the latest V2 MicroPython, the latest release for micro:bit V1 can be found here: https://github.com/bbcmicrobit/micropython/releases/tag/v1.0.1

At the moment uFlash attaches the user code to the hex file using an old method that will not work with V2. This method is somewhat deprecated, as you only get 8 KBs of flash space and has other limitations. The best way to add files to the micro:bit is to send them via serial with libraries like https://github.com/ntoll/microfs (other boards also use things like https://github.com/scientifichackers/ampy).
It's possible uFlash will be updated with a universal hex, but in the meantime uFlash also offer the option to flash any hex file: https://github.com/ntoll/uflash/blob/dc04cf2134b877c03eb54180129b42f4b8a9e546/uflash.py#L458

A related issue in this area is https://github.com/vlasovskikh/intellij-micropython/issues/29.

vlasovskikh, 2020-12-12:
@microbit-carlos Thanks for the update! I already use ampy for other MicroPython devices, I might consider changing the upload mechanism for Micro:bit as well.

microbit-carlos, 2020-12-14:
Awesome! Let me know if you hit any issues.

microbit-carlos, 2020-12-14:
Actually we've recently found a small issue in the V2 port with the serial getting some random data in some occasions, which can affect this. We are currently looking into this and plan to address it for the next beta release, which will be published in https://github.com/microbit-foundation/micropython-microbit-v2/releases/ as beta.3.

microbit-carlos, 2020-12-23:
Beta 3 has been released, which should solve he serial issues: https://github.com/microbit-foundation/micropython-microbit-v2/releases/tag/v2.0.0-beta.3

regalme, 2020-12-23:
Awesome I will check it out and validate after the holidays!