JetBrains/intellij-micropython #128: Bug with flashing files

berkut0, 2020-12-08:
Issue with uploading files included in subfolders — doubled backslash with following 'b' character transforms into special symbol (in my case)

<details>
  <summary>Images</summary>
  
[image]
[image]
[image]
</details>

maro48m, 2021-02-12:
i've had the same error with MicroWebServer2 - upload of urlUtils.py crashed
All this is due to path replacement in microupload.py:
in make_dirs method path separator is replaced to '/' in line 96:
`posix_path = path.replace(os.path.sep, '/')`
but in main upload method it's not happening: lines 76 to 79:

```
if remote_dir:
      make_dirs(files, remote_dir, created_cache)
  with open(local_path, 'rb') as fd:
      files.put(remote_path, fd.read())
```
if you replace path separator in remote_path file upload will work correct:
`files.put(remote_path.replace(os.path.sep, '/'), fd.read())`

Alienmario, 2021-02-26:
Having the same issue (ESP32).

PR #144 (2021-04-16): Fixed Upload Path bug
Starting from a blank file system, files can upload with the filename '//path//to/file.py' in the root directory instead of being uploaded into the directory 'path/to/'. This PR changes the remote_path to a posix path like make_dirs does to ensure the file is uploaded properly regardless of the OS used.

EDIT: This Fixes #128
closed 2021-04-16 by lensvol