JetBrains/intellij-micropython #131: what's the device path on windows system?

Alice-and-Bob, 2021-01-19:
I saw your example.
You appear to be running PyCharm on a Linux system, and your ESP8266 chip is also connected to a Linux system.
However, I use a Microsoft Windows operating system, so I'm not sure what the device path required in the plugin is.
You know, on Linux the chip might behave like a file, but on Windows it doesn't.
So what values should I put in the device path options?

jdjjm, 2021-01-20:
It will be a COM port. If you search for `Device Manager` in the start menu, you can view the connected devices. They should show up under `Ports (COM & LPT)` and will look something like `USB Serial Port (COM4)`. In this case you would use COM4 as your device path.

Alice-and-Bob, 2021-01-20:
> It will be a COM port. If you search for `Device Manager` in the start menu, you can view the connected devices. They should show up under `Ports (COM & LPT)` and will look something like `USB Serial Port (COM4)`. In this case you would use COM4 as your device path.

Thanks for your reply! I have solved this question.

closed 2021-01-20 by Alice-and-Bob