JetBrains/intellij-micropython #132: Enhancement - add Raspberry Pi Pico microcontroller support

smith9, 2021-01-21:
Hi,

Any plans to support the new Raspberry PI Pico microcontroller board?

vlasovskikh, 2021-01-21:
I don't have this device, so I cannot test it at this point. Contributions adding Raspberry Pi Pico support to the plugin are welcome. The plugin supports several devices and the code for supporting a new one is relatively easy to add.

lwr20, 2021-01-22:
I just received my Raspberry Pi Pico board a few minutes ago, but I'm new to this.  Is there a doc somewhere that demonstrates how to add a new device?  Or a PR that added a different device that I can use as an example?

lwr20, 2021-01-22:
Hah - I just found this: https://github.com/vlasovskikh/intellij-micropython/pull/114  
I'll see if I can knock up something similar.

tinue, 2021-01-29:
It is working quite ok with the "Pyboard" device. The only exception is the soft reset: For whatever reason this is not working on my Mac. It is sending the right commands (Ctrl-C, Ctrl-D), but the board does not perform the reset.
If I connect to the board after having flashed through IDEA, with e.g. screen, it immediately runs the (pending?) reset. Maybe it's just a timing issue, e.g. disconnecting too quickly.
To be clear: I am using IDEA with Python and MicroPython plugins, not PyCharm.

mlewus, 2021-02-03:
I'm not familiar enough with Python to help code it, but it would be great to have Pi pico support in PyCharm :) I will try to test it using the "Pyboard" device re: @tinue

cakoronkowski, 2021-02-12:
> I don't have this device, so I cannot test it at this point. Contributions adding Raspberry Pi Pico support to the plugin are welcome. The plugin supports several devices and the code for supporting a new one is relatively easy to add.

@vlasovskikh would mailing you one of these devices be helpful for this? I have a spare and they are very inexpensive in my area

mlewus, 2021-02-12:
I live in New Jersey, USA. I also have a known good Pico board I could mail you. Let me know.

smith9, 2021-02-24:
If someone like timsavage does a pull request, what does it take to get that merged in - as in what is the process/timeline?
Does it need other people to fork the plugin and pull in the pull request to do testing?

vlasovskikh, 2021-08-05:
Raspberry Pi Pico support is coming to the MicroPython plugin for PyCharm v1.2:

* Ability to flush your files to RPi Pico via MicroPython run configurations
* Basic type hints (please contribute more of them via PRs!)

🙏 Thank you @timsavage for contributing code for Raspberry Pi Pico support! Thanks @lensvol for reviewing the PR!

closed 2021-08-05 by vlasovskikh