JetBrains/intellij-micropython #134: Unable to add MicroPython facet in IntelliJ IDEA 2020.3.1

TwoSlick, 2021-01-23:
In a project set up with a python module, the MicroPython option is not visible for adding a facet.

[image]

IDEA details:
IntelliJ IDEA 2020.3.1 (Ultimate Edition)
Build #IU-203.6682.168, built on December 29, 2020
Runtime version: 11.0.9.1+11-b1145.63 x86_64
VM: OpenJDK 64-Bit Server VM by JetBrains s.r.o.
macOS 10.16

brunnels, 2021-02-10:
I can confirm the same on windows and linux.

If I add the facet to the project in pycharm and then open the project in intellij then everything compiles and uploads correctly.  I think it's just a setting issue where it's not showing up under facets or under Settings -> Languages and Frameworks

DFayro, 2021-02-27:
Can confirm the same on intellij 2020.3.* and 2020.2.4

jphalip, 2021-04-28:
I'm having the same issue with Intellij Ultimate 2021.1.

lucavallin, 2021-07-11:
Same problem with Intellij Ultimate 2021.1 on macOS.

vranisimo, 2021-07-14:
The same issue on
IntelliJ IDEA 2021.1.3 (Community Edition)
Build #IC-211.7628.21, built on June 30, 2021
Runtime version: 11.0.11+9-b1341.60 amd64
VM: OpenJDK 64-Bit Server VM by JetBrains s.r.o.
Windows 10 10.0

vlasovskikh, 2021-08-05:
Are you sure you have Python modules in your Modules section of the Project Structure? You cannot add a MicroPython facet for a non-Python module, even for a non-Python module with a Python facet configured for it. I've checked it on IntelliJ 2021.2 and the MicroPython facet becomes available for me if I have at least one Python module in my Modules.

singularcitrus, 2021-11-11:
IDEA Ultimate 2021.2.1 with python3.9 still not working

vlasovskikh, 2021-12-01:
@singularcitrus It look like `pico` is a non-python module with a Python facet added. Could you please select the `pico` module itself in the first screenshot and check what its SDK is?

I'd like to point to my comment above about the inability to add a MicroPython facet to a non-Python module, even if this module has a Python facet enabled for it:

> Are you sure you have Python modules in your Modules section of the Project Structure? You cannot add a MicroPython facet for a non-Python module, even for a non-Python module with a Python facet [...]

singularcitrus, 2021-12-01:
Thanks for your reply, I did indeed add the module wrong. My lovely IDE kept selecting "pico" (which is the project name) when trying to add the module, after some clicking away and trying to add it again everything works!

I appreciate the help

vlasovskikh, 2021-12-01:
@singularcitrus I'm glad it helped :) I'm closing this issue as answered. I agree that the user experience with Java modules vs Python modules + Python facet is confusing.

closed 2021-12-01 by vlasovskikh