JetBrains/intellij-micropython #136: Can't flash Python files with doctests inside

lensvol, 2021-02-02:
**Software information:**
* PyCharm 2020.3.3 (Professional Edition)
* MicroPython plugin v. 1.1.3

**Steps to recreate:**
1. Create a Python file (name does not matter)
2. Put following code inside:
```python

class SimpleClassWithDoctests:
    """
    Hello, world!

    >>> from machine import Pin, SPI
    >>> spi = SPI(mosi=Pin(13), sck=Pin(14))
    """
    pass
```
3. Right-click it in the "Project" file tree

**Expected**: 
* "Flash <filename>..." option in the context menu

**Actual**:
* "Run Doctests in <filename>" option is in the context menu

jdjjm, 2021-02-03:
Just create a new run configuration. Click on the configuration dropdown and choose `Edit Configurations...`. Click the `+` in the top left and choose `Micropython` from the list. Give the configuration a name and put in the full file path for the python file you are trying to upload. Click `OK` and you should be able to flash the file. Once a configuration is made you can right click the file and it should now give you the option to Flash instead of running doctests. 

What is the purpose of the doctests anyway? I am not aware that they were available in micropython.

lensvol, 2021-02-04:
They came in with an external module that I use, and technically nothing prevents you from using them with MicroPython.

Thank you, I will use the workaround. In the future there may be a space for discussion about moving away from (ab)using Run Configurations to a standalone execution mechanism with dedicated menu options.