JetBrains/intellij-micropython #137: Can't import uos in PyCharm

quaintdev, 2021-02-04:
PyCharm keeps telling me that there is no module named uos but I can import it on esp8266 without any issue. Any suggestions?

Wanyor, 2021-03-01:
You can use this tool to generate pyi stubs: [https://github.com/Josverl/micropython-stubber](url) , and place the stubs at C:\Users\Administrator\AppData\Roaming\JetBrains\PyCharmCE2020.3\plugins\intellij-micropython\typehints.

Wanyor, 2021-03-01:
> PyCharm keeps telling me that there is no module named uos but I can import it on esp8266 without any issue. Any suggestions?

You can use this tool to generate pyi stubs: [micropython-stubber](https://github.com/Josverl/micropython-stubber) , and place the stubs at C:\Users\Administrator\AppData\Roaming\JetBrains\PyCharmCE2020.3\plugins\intellij-micropython\typehints.

hlovatt, 2021-08-08:
it's on my list to create for https://github.com/hlovatt/PyBoardTypeshed and then hopefully it will be included in the Micropython plugin (many of the existing typesheds/stubs in Micropython come from this project). Give me a few weeks :).

vlasovskikh, 2021-12-01:
Should be fixed by #176.

closed 2021-12-01 by vlasovskikh