JetBrains/intellij-micropython #138: Micropython REPL error with pyserial 3.5

d-marquina, 2021-02-07:
Hi, I recently started tinkering with Micropython, ESP32 and Windows 10. Everything works perfectly except for the Micropython REPL: backspace and arrows are throwing strange characters.

I found out that this was an old issue regarding miniterm and pyserial 3.4, already resolved and closed. However, pyserial 3.5 was released last year and the problem came back.

The fastest solution was to downgrade pyserial to 3.4 but I wanted to report this behaviour.

krzim, 2021-02-10:
I can confirm that I'm experiencing the same issue on Windows 10 with an ESP8266 and US keyboard. The downgrade to pyserial 3.4 resolves the issue.

vlasovskikh, 2021-12-01:
It should be fixed by #175.

closed 2021-12-01 by vlasovskikh