JetBrains/intellij-micropython #145: Upload to ESP32 failed

dmp2tech, 2021-03-08:
Hello,
I have a problem uploading to my ESP32. When I try to upload with pycharm I have an error: ```ampy.pyboard.PyboardError: could not enter raw repl ```
So I tried to upload directly from the terminal without a virtual environment. And it works. So I logged into the virtual environment, updated ampy to version 1.1.1 and was able to upload it from the terminal. On the other hand, I cannot upload with pycharm because it wants to update version 1.0.5. How do I get the extension to accept version 1.1.1 of ampy. I'm on big macos on.

Arcadia197, 2021-04-12:
Have you tried to change File -> Settings -> Languages&Frameworks -> MicroPython -> Device type to esp8266 in PyCharm?

As ampy is giving a pyboard error maybe you have forgotten to set this.

While esp32 is not directly supported it should work for most cases with the esp8266 settings (atleast it does for me)
Just watch out if you want to include folder structures - this did not work for me

lensvol, 2021-04-12:
This can be related to the issue solved here: https://github.com/vlasovskikh/intellij-micropython/pull/142

You can try to increase hardcoded delay on the following line and see if it helps: https://github.com/vlasovskikh/intellij-micropython/blob/0322fb7c50bed3e196ee253fbb0f613344a56a48/scripts/microupload.py#L121

Path to `microupload.py` can be found in the _Terminal_ window that appears when you try to upload files.

kpostekk, 2021-05-26:
> You can try to increase hardcoded delay on the following line and see if it helps

Solution by @lensvol has worked for me. I have changed value from `0.5` to `2.5` and it works.

[image][image]

_You can open file by following command_ 
`code "~/Library/Application Support/JetBrains/PyCharm2021.1/plugins/intellij-micropython/scripts/microupload.py"`

jwoolaway, 2021-08-23:
Running:
OSx 11.5.2
PyCharm 2021.2 V 11.0.11+9-b1504.13 x86_64
Plug-in MicroPython 1.2
The described edit to:
`"/Users/<USER_NAME>/Library/Application Support/JetBrains/PyCharm2021.2/plugins/intellij-micropython/scripts/microupload.py"`
Fixed it.

jadn123, 2022-01-09:
Fix works for me as well. Can this be considered being fixed in a PR?

edwinm, 2022-08-21:
It's August 21st 2022 and I still had to use this hack to get my file uploading to work.

I'm using a five days old build.

PyCharm 2022.2.1 (Community Edition)
Build #PC-222.3739.56, built on August 16, 2022

Dear JetBrains, please fix this.

Elizaveta239, 2022-09-02:
Hi! We'll try to include it in the next update.

Dans-Droids, 2022-11-28:
> Running: OSx 11.5.2 PyCharm 2021.2 V 11.0.11+9-b1504.13 x86_64 Plug-in MicroPython 1.2 The described edit to: `"/Users/<USER_NAME>/Library/Application Support/JetBrains/PyCharm2021.2/plugins/intellij-micropython/scripts/microupload.py"` Fixed it.

I just tried editing this file on Windows (Program Files\JetBrains\PyCharm.....\ etc) but there is no plugins\intellij-micropython\ folder.. I have downloaded the MicroPython Plugin from the Plugin Store, am I missing something?!

kpostekk, 2022-11-28:
Have you checked app data folder?

Dans-Droids, 2022-11-28:
> Have you checked app data folder?

ah Thank you i didnt think of checking there...
In Users\AppData\JetBrains\...\plugins\  there is only ```intellij-micropython.zip``` in the plugins folder, should I extract it to the Program Files\JetBrains\...\plugins\ folder?

LouisCAD, 2022-11-28:
@Elizaveta239 When will the next update be?

jaylinwylie, 2024-03-19:
Are we still to do this hack? Kind of baffling that the ESP32 isnt already in the list.

ygorgasparin, 2024-09-07:
Here works as well