JetBrains/intellij-micropython #147: Flashing to a ESP8266 12E isnt working.

solarwoof13, 2021-03-26:
I am trying to connect a ESP8266 and I am having issues, my computer sees the device, I have flashed the devices and updated all firmware. But when I address it in the Run section it says it doesnt recognize the device. I have watched several YT videos where people have success. I must be missing a setting that I cannot find. Please help | Solarharvey79Mar 26, 2021, 15:31 GMT+1 I am trying to connect a ESP8266 and I am having issues, my computer sees the device, I have flashed the devices and updated all firmware. But when I address it in the Run section it says it doesnt recognize the device. I have watched several YT videos where people have success. I must be missing a setting that I cannot find. Please help
[image]
[image]

vlasovskikh, 2021-03-26:
"Path" in the run configuration is the path to the folder or file you want to flash to your MicroPython device. You should enable MicroPython support for your project in "Preferences -> Languages and Frameworks -> MicroPython". Make sure you select "ESP8266" as your device type. You can use either "Auto-detect device path" or enter "Device path" manually there.

Does it solve your problem?

solarwoof13, 2021-03-29:
> "Path" in the run configuration is the path to the folder or file you want to flash to your MicroPython device. You should enable MicroPython support for your project in "Preferences -> Languages [...]

I did this multiple times, I just checked and it is still set to ESP8266, but will not auto detect the device.

vlasovskikh, 2021-03-29:
@solarwoof13 What are the vendor ID and device ID of the USB serial controller of your MicroPython device?

You could try to run https://github.com/vlasovskikh/intellij-micropython/blob/master/scripts/findusb.py to scan the IDs of connected USB devices. Make sure you do it in the virtualenv of your project with docent and pyserial packages installed.

solarwoof13, 2021-03-30:
> Product ID: 0xea60
          Vendor ID: 0x10c4  (Silicon Laboratories, Inc.)
          Version: 1.00
          Serial Number: 0001
          Speed: Up to 12 Mb/s
          Manufacturer: Silicon Labs
          Location ID: 0x14600000 / 38
          Current Available (mA): 500
          Current Required (mA): 100
          Extra Operating Current (mA): 0

I tried it and that is also not working. My pycharm isn't locating the pyserial package in the settings.

solarwoof13, 2021-03-30:
Ok. I had my setting in conda not python for the vertenv. I will try now.

solarwoof13, 2021-03-30:
This is what it returns. 

> /usr/local/bin/python3.9 /Users/solar_mike/PycharmProjects/find_usb/main.py
Usage:
    findusb [VID:PID...]
Arguments:
    VID:PID       Vendor ID and Product ID for your USB device.
If you run it with no arguments, it outputs the list of all the USB devices it
has managed to find.

Process finished with exit code 1

vlasovskikh, 2021-03-30:
Somehow command line parsing didn't work in your run. Please read the script and try to list your serial ports by yourself.

solarwoof13, 2021-03-30:
I still cannot figure it out. I hate that I am paying for PyCharm but am not going to use it as I got it to work in VSCode. They should be paying me to post this. I got it to work in 2 minutes using these instructions--> https://opensource.com/article/20/7/python-rt-thread

vlasovskikh, 2021-03-30:
@solarwoof13 The intellij-micropython plugin is open-source and it's basically my personal 20% project, as stated in the README.

closed 2021-03-30 by vlasovskikh