JetBrains/intellij-micropython #161: I couldn't make work esp32 device after upload program files from Pycharm  to esp32 module

electronic4706, 2021-04-12:
Hello,
I am trying to upload and run my esp32 device with Pycharm but it is not working .
did implementation :
I have installed Pycharm end updated it. I installed Micropython plugin and I can connect to esp32 module.
I am uploading my program files with Pycharm after I install micropython esp32 firmware with uPyCraft.
however, device not wake up not working. Why? what is the wrong?
Whereas, esp32 device can wakeup and work  if I install same firmware and upload same files to esp32 module with uPyCraft. 
firmware : esp32-idf3-20210202-v1.14.bin
uploaded files: https://github.com/Makerfabs/Project_Soil-Moisture-Sensor-Wifi/tree/master/LoraS2G/workSpace

I couldn't make work esp32 device after upload program files from Pycharm  to esp32 module

jdjjm, 2021-04-12:
Have you enabled the micropython for the project under `Settings -> Languages and Frameworks -> Micropython -> Enable Micropython Support` and chosen ESP8266 as the device type. You will also need to put the port the device is connected to under `device path`

Can you connect to the REPL by doing `Tools -> MicroPython -> MicroPython REPL`?

Start by trying to upload a single file by creating a new micropython run configuration or by right clicking the file and choosing `Run Flash <filename>.py`.

Let us know how this goes.
If this doesn't work please share some more information about what isn't working. Please tell us if you encountered an error or at what point did it fail etc. This should help us all figure out what is going on much easier. Thanks

electronic4706, 2021-04-13:
> Have you enabled the micropython for the project under `Settings -> Languages and Frameworks -> Micropython -> Enable Micropython Support` and chosen ESP8266 as the device type. You will also need to put the port the device is connected to under `device path`
> **Hi , firstly thank you for the your relpy. sorry about if ı am late to answer. I can do this  paragraph**

> Can you connect to the REPL by doing `Tools -> MicroPython -> MicroPython REPL`?
> **_I can do this  paragraph_**

> Start by trying to upload a single file by creating a new micropython run configuration or by right clicking the file and choosing `Run Flash <filename>.py`.
**_> **_Actually sory I didn't understand "by creating a new micropython run configuration " however, ı tried step by step :  I created new project . automatically it made main.py file. After that , i have enabled the micropython for the project under `Settings -> Languages and Frameworks -> Micropython -> Enable Micropython Support` and chosen ESP8266 as the device type. I putted the port the device is connected to under `device path`  . I opened tools/Micropython/MicropythonREPL. ireseted device by button on device . it enswered by check process in terminal. when i tryed  Run flash mailn.py, it turned answer "ampy.pyboard.PyboardError: failed to access COM5
" . I need files that I shared. I did upload them yet as I sent one by one those files. They achieved to upload._** **_When opening the files with PyCharm, I compared them with my original project files to see if something changes. but he did not make a change._**_**

> Let us know how this goes.
> **_ı didn't achieve it_** 

> If this doesn't work please share some more information about what isn't working. Please tell us if you encountered an error or at what point did it fail etc. This should help us all figure out what is going on much easier. Thanks
>**_"ampy.pyboard.PyboardError: failed to access COM5
"_**

electronic4706, 2021-04-13:
> Have you enabled the micropython for the project under `Settings -> Languages and Frameworks -> Micropython -> Enable Micropython Support` and chosen ESP8266 as the device type. You will also need to put the port the device is connected to under `device path`
> **_Hi , firstly thank you for the your relpy. sorry about if ı am late to answer. I have been do this  paragraph_**
> Can you connect to the REPL by doing `Tools -> MicroPython -> MicroPython REPL`?
>  **_I have been can do this  paragraph_**
> Start by trying to upload a single file by creating a new micropython run configuration or by right clicking the file and choosing `Run Flash <filename>.py`.
> _**Actually sory I didn't understand "by creating a new micropython run configuration " however, ı tried step by step :  I created new project . automatically it made main.py file. After that , i have enabled the micropython for the project under `Settings -> Languages and Frameworks -> Micropython -> Enable Micropython Support` and chosen ESP8266 as the device type. I putted the port the device is connected to under `device path`  . I opened tools/Micropython/MicropythonREPL. ireseted device by button on device . it answered by checking process in terminal. when i tryed  Run flash mailn.py, it turned answer "ampy.pyboard.PyboardError: failed to access COM5
" . I need files that I shared. I did upload them yet as I sent one by one those files. They achieved to upload._** **_When opening the files with PyCharm, I compared them with my original project files to see if something changes. but he did not make a change.**_
> Let us know how this goes.
>**_sorry , I didn't achieve it_**
> If this doesn't work please share some more information about what isn't working. Please tell us if you encountered an error or at what point did it fail etc. This should help us all figure out what is going on much easier. Thanks
>_**"ampy.pyboard.PyboardError: failed to access COM5"**_

jdjjm, 2021-04-13:
Make sure the REPL is closed before you upload code. You cannot flash a file if a REPL connection is active because they use the same serial port.

If this is not the issue try running the following in the Python Console
```
import serial
serial.Serial('COM5', baudrate=115200)
```
This should show the actual error you are encountering. Let us know what this is.

electronic4706, 2021-04-14:
> Make sure the REPL is closed before you upload code. You cannot flash a file if a REPL connection is active because they use the same serial port
> **_thank you very much but evenelse ı upload ı couldn't running device because last time i sent files . then i restart device . display didn't show anything on screen. i must show assigned up and subnet musk ._** 
> If this is not the issue try running the following in the Python Console
> 
> ```
> import serial
> serial.Serial('COM5', baudrate=115200)
> ```
> i will check it
> This should show the actual error you are encountering. Let us know what this is.

electronic4706, 2021-04-14:
I sent all files by give message for each in follow:
**_Connecting to COM5
Uploading files: 0% (0/1)
C:\Users\tunc\Desktop\pycharm lora\workSpace\webserver.py -> workSpace\webserver.py
Uploading files: 100% (1/1)
Soft reboot_**
Afterwards, I do Tool/Micropython/MicropythonREPL for listining device. So, I reset device. I am giving message in follow:
**_Device path COM5
Quit: Ctrl+] | Stop program: Ctrl+C | Reset: Ctrl+D
Type 'help()' (without the quotes) then press ENTER.
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5008
ho 0 tail 12 room 4
load:0x40078000,len:10600
ho 0 tail 12 room 4
load:0x40080400,len:5684
entry 0x400806bc
MicroPython v1.14 on 2021-02-02; ESP32 module with ESP32
Type "help()" for more information._**
However Device not working if i try to sent through Pycharm...

jdjjm, 2021-04-15:
Can you open the REPL and run
```
import os
os.listdir()
```
And let me know the result.

electronic4706, 2021-04-15:
> import os
> os.listdir()
it show in following when i write that :
**_['workSpace\test.py', 'workSpace\\config_lora.py', 'workSpace\\controller.py', 'workSpace\\controller_esp.py', 'workSpace\\controller_esp_lora_oled.py', 'w
orkSpace\\display_ssd1306_i2c.py', 'workSpace\\lora_node.py', 'workSpace\\main.py', 'workSpace\\ssd1306.py', 'workSpace\\sx127x.py', 'workSpace\\webserver.
py', 'workSpace']_**

it is the result that must be. amount of files is correct but it isn't working...

jdjjm, 2021-04-16:
I think you might be encoutering the bug #128. This is where a file is uploaded to the root directory with the a name of _"foldername/filename.py"_ instead of _"filename.py"_ inside the _foldername_ directory.
PR #144 fixes this so it should be fixed in a future release. 
For now you will need to edit the _microupload.py_ file and replace line 79 `files.put(remote_path, fd.read())`  with `files.put(remote_path.replace(os.path.sep, '/'), fd.read())`
If you need to know where your _microupload.py_ file is located you can look at the second element of the first line when you flash a file. It will be the path to your microupload file.

After you have fixed this run `Tools -> MicroPython -> Remove All Files from MicroPython Device`.
Once you have cleared the files, reupload them like before and then run `os.listdir()` again and verify all the files are inside the workSpace folder. You can also do `os.listdir('workSpace')` to list the files inside the workSpace folder once its fixed.

electronic4706, 2021-04-16:
I changed line. After that , I see just file names not "workspace" directory name in path. it showed just files name. However, it isn't work still. So, I tried some thing. for example. I tried  create new project and ı copied my files to its directory folder. after that, I **run "flash files"** .  it didn't work. I realized .idea folder had uploaded. I removed all files in device with   **Tools -> MicroPython -> Remove All Files from MicroPython Device**. before Run flash files ago, I deleted **.idea folder** in project directory. I did **Run flash files** for all files again. After that, I reset device. Now device working :). You has been right. it related with directory. it shown working now. I didn't try lora network but it connected to router and it took IP form DHCP server. I'm guessing that files working correctly. 
Thank you very much helping . I can answer gladly, If you wonder anything about I tried way.

jdjjm, 2021-04-16:
Glad to hear you got it working.

electronic4706, 2021-04-16:
I'm a rookie on Github.  I guess when users find a solution, they close the issue. Isn't it ?  thank you again for everything.

jdjjm, 2021-04-16:
Yes close the issue when it is solved. Thanks

electronic4706, 2021-04-16:
thanks . stay healthy.

closed 2021-05-05 by electronic4706