JetBrains/intellij-micropython #166: I can't send py files to ESP-01 by PyCharm  or uPyCraft. I need a guy who programmed ESP-01 with micropython language.

electronic4706, 2021-05-05:
Hello. dear productive friends. I need a guy who programmed ESP-01 with micropython language.  
 I have got 1 blue led version product. I am tries programming with micropython by PyCharm IDE.  I think that I flashed it with esp8266-512k-20190125-v1.10.bin by uPyCraft . however, connection is not successful, so couldn't send my files to it . upyCraft can't connect it . 
I am using MAKERFABS  CP2104 USB2UART Module. connection :
CP2104    ESP01
rx      ->       tx
tx      ->       rx
VIO   ->      3v3
DTR  ->      Rst
                   EN -> 3v3
                   DO0 -> GND
GND ->      GND
I flashed by upyCraft (Maybe I think so.) but i didn't send py files and i can't connection to ESP-01. 
I tried PyCharm. I saw com port but i didn't send files. 
Can any proactive guy help me, please ?

Wanyor, 2021-06-06:
My SerialPort settings， you can try：

baudrate：115200

like this:
[image]

electronic4706, 2021-06-22:
:) :) :)

closed 2021-06-29 by electronic4706