JetBrains/intellij-micropython #169: Please reopen closed issue 'Unable to run any micropython in PyCharm for ESP8266 #120'

hcet14, 2021-07-01:
Please try to fix [https://github.com/vlasovskikh/intellij-micropython/issues/120#issuecomment-872582972](url) it shouldn't have been closed. I started with pycharm and the micropython plugin a couple of days ago. Great stuff. Keep it up! I'm unable to solve the problem and couldn't find any information which helps. I'm not a programmer. Please see also [https://forum.micropython.org/viewtopic.php?f=15&t=10771](url)

jdjjm, 2021-07-05:
That issue does not need to be reopened, it has been solved.
You are doing the same thing that AstroDabb was. You are running a micropython script with regular python instead of uploading it to a microcontroller.
Once you have enabled micropython support (I assume you have since you can access the REPL), you need to right click your main.py and choose `Run Flash main.py`. 

You should familiarise yourself with the platform you are using
[This is the documentation for Run Configurations in PyCharm](https://www.jetbrains.com/help/pycharm/run-debug-configuration.html)
Keep in mind with micropython, you are not running the code directly, instead you are running an upload script to put your code on your microcontroller.

vlasovskikh, 2021-08-05:
@hcet14 The problem you have (`PyboardError: failed to access COM3`) is not related to the original problem of issue #120. That issue was about some confusion with which run configuration to use: Python or MicroPython.

I'll update this issue so it reflects the issue you have if you don't mind.

vlasovskikh, 2021-08-05:
@hcet14 I don't have a Pyboard device and a Windows machine at hand, so I cannot reproduce your issue. The MicroPython plugin actually tells the exact command it runs to flash files to your device. It's the first line in the Run tool window:

```
...\python.exe ...\microupload.py [parameters...]
```

You might experiment with running microupload.py and try to debug the issue.

To other Pyboard users: any help on reproducing and fixing the issue is appreciated.

jdjjm, 2021-08-05:
@vlasovskikh I do not believe what @hcet14 is seeing is a bug. Take a look at the micropython.com [Forum post](https://forum.micropython.org/viewtopic.php?f=15&t=10771)  he made. It shows some pictures and you can see that he is running the script with pure python, not flashing it to the microcontroller. He just need to change his run configurations and I think he will be good to go.

vlasovskikh, 2021-08-05:
@jdjjm In the comment by @hcet14 in #120 (July 2) they show a screenshot where they run a MicroPython run configuration: it's microupload.py with the corresponding parameters.

jdjjm, 2021-08-05:
@vlasovskikh Woops, I missed those screenshots. Looks like this might be a bug.