JetBrains/intellij-micropython #179: [Feature] Extract board stubs to pypi packages

mrkeuz, 2021-11-10:
According [PEP-0561](https://www.python.org/dev/peps/pep-0561/), stubs can be distributed via pip.

It would be good to publish stub packages to pip for easy installation just via `pip`. There is another team that tried to support similar versions of stubs [micropython-stubs](https://github.com/Josverl/micropython-stubs) but in auto-generated manner.

I have created similar issue [#123](https://github.com/Josverl/micropython-stubs/issues/123). It would be good to join forces and extract stubs to common project, for ability deploying them to PyPi and support by community together. It brings ability to use this stubs in linters, and other tools. Also, this allows for involving more community contributions, in my humble opinion.

Sincerely to JetBrains team and contributors! 🚀

vlasovskikh, 2021-12-01:
I like this idea. We can modify the plugin to require installing these stubs for your device type similarly to what we already do for dependencies for communication with your device.

See https://github.com/vlasovskikh/intellij-micropython/blob/a068a81922151686142d01d93d4375c836ed92e6/src/main/kotlin/com/jetbrains/micropython/devices/PyboardDeviceProvider.kt#L50 as an example.

vlasovskikh, 2021-12-01:
cc @hlovatt you might be interested in this as a stubs contributor.

hlovatt, 2022-01-04:
Sorry it took me so long to respond to this. It's a great idea. Took me a while battling PyPI to get something that works!

You  can now install from PyPI:

```bash
    pip install --upgrade micropython-typesheds
    python -m micropython-typesheds <destination-directory>
```

Josverl, 2023-03-20:
for recent stubs : https://github.com/Josverl/micropython-stubs or [Micropython stub packages on PyPI](https://pypi.org/search/?q=-stubs&o=&c=Programming+Language+%3A%3A+Python+%3A%3A+Implementation+%3A%3A+MicroPython)