JetBrains/intellij-micropython #189: It's not possible to flash and use REPL

berkut0, 2022-04-10:
Hi

I have been using Sublime with the [UPIOT ](https://github.com/gepd/uPiotMicroPythonTool)plugin a lot lately to work with micropython. It still seems to me to be the most reliable tool for working with micropython prototypes. I have a little note about installation and use [in my repository](https://github.com/BrightestBrains/LightBrain/tree/main/code/micropython).

One of the most important features of UPIOT is the seamless use of the COM port. I don't need to disable the REPL console to use sampy / upload files. It works in the same console. Using this plugin, when flashing with the console open, an error pops up - COM port, of course, busy. This feature just wasn't provided? Is this feature available because of modified ampy — sampy? Is it possible to implement this seamless behavior in intellij plugin?

This UX doesn't feel right — the terminal window is just in one window, the error in another, close the console, then flash, then open...

berkut0, 2022-04-11:
I think, pull request #139 related

robomartin, 2022-06-13:
My current work-around is to have Thonny open and connected to the board for upload and REPL.  I edit the code on PyCharm and never connect to the board from it.  I would very much like to not have to do this and work entirely from PyCharm.

ssddanbrown, 2022-11-21:
I've just spent some time attempting to replicate this kind of workflow within pycharm. Unfortunately I could not get it down to a button press, but I got it down to a terminal line run, although you need to quit out of the terminal REPL between runs.
Here's my approach in case if helps anyone else:

1. Within my local pycham venv I installed the `rshell` package.
2.  I created a `flash_dev.sh` file with the below contents. Make it executable (`chmod +x flash_dev.sh`)

```shell
#!/bin/bash
rshell --buffer-size=512 --quiet "cp $1 /pyboard/dev.py; repl ~ import dev"
```

3. You can then use a pycharm terminal window to run the script, passing the file you want to flash like: `./flash_dev.sh script.py`.
4. You can exit out the repl with `Ctrl+X`. A re-run from being in the repl would just need a `Ctrl+X` followed by :arrow_up: then `Enter` in the terminal.

I have other `flash_*.sh` scripts in [my working repo here](https://github.com/ssddanbrown/raspi-pico-playground/tree/50dd685c21c35a8fe1df2d9a8a8f15bffea79724) to flash files across (without running) and to flash a file as the `main.py` file on the device.

I tried to wrap this in a run configuration but failed since you can't pass the active filename to the shell run configurations. You could hard-code it to a specific script, but then running without remembering to close the existing repl session makes things messy.