JetBrains/intellij-micropython #193: 插件 无法上传esp8266开发板上

CuiHaiPeng, 2022-06-11:
您好 我使用此插件 无法上传到esp8266开发板上 

报错信息

```shell
/usr/local/bin/python3.9 "/Users/cuihaipeng/Library/Application Support/JetBrains/PyCharmCE2022.1/plugins/intellij-micropython/scripts/microupload.py" -C /Users/cuihaipeng/PycharmProjects/mircoPy -v /dev/cu.usbserial-14130 /Users/cuihaipeng/PycharmProjects/mircoPy/main.py
Connecting to /dev/cu.usbserial-14130
Uploading files: 0% (0/1)
/Users/cuihaipeng/PycharmProjects/mircoPy/main.py -> main.py
b'rw'
Traceback (most recent call last):
  File "/Users/cuihaipeng/Library/Application Support/JetBrains/PyCharmCE2022.1/plugins/intellij-micropython/scripts/microupload.py", line 139, in <module>
    main(sys.argv[1:])
  File "/Users/cuihaipeng/Library/Application Support/JetBrains/PyCharmCE2022.1/plugins/intellij-micropython/scripts/microupload.py", line 79, in main
    files.put(remote_path, fd.read())
  File "/usr/local/lib/python3.9/site-packages/ampy/files.py", line 208, in put
    self._pyboard.enter_raw_repl()
  File "/usr/local/lib/python3.9/site-packages/ampy/pyboard.py", line 192, in enter_raw_repl
    raise PyboardError('could not enter raw repl')
ampy.pyboard.PyboardError: could not enter raw repl
```

项目基本信息
运行配置

CuiHaiPeng, 2022-06-11:
在thonny上面 是可以上传 并且运行代码的 但是在idea上面就不可以

hashming, 2022-11-13:
出现这个错误可以重新插拔一下usb，就不会出现这个错误了

jokerElsa, 2023-02-15:
https://github.com/scientifichackers/ampy/issues/19
I found this could help if you suffering the "could not enter raw repl" error
这里面往下翻，指出了一个函数要sleep 2秒解决，我用这个修复好了。如果修复好了可以关issue了
我是上传之后点不起灯，共勉吧