JetBrains/intellij-micropython #199: Feature request: option to close REPL terminal on run/upload

wz2b, 2022-07-02:
This plugin is awesome!  I am using it on an ESP-WROOM32 module, and it works perfectly.

I would like to request one feature that would help my workflow.  I'd love an option that would close the REPL terminal when you pushed RUN, because my sequence is RUN (upload) then go to the REPL and try things out.  It's not a huge deal to have to close the REPL terminal every time, but it would save some clicking if it could either close the REPL or, even better, pause it so that it can steal the serial port for the upload.

vekunz, 2022-07-09:
I think that this would be solved by this PR https://github.com/vlasovskikh/intellij-micropython/pull/139

elmot, 2024-07-24:
I believe this is fixed with "Stop MicroPython REPL" before launch task
[image]