JetBrains/intellij-micropython #200: Installing modules throws errors

noises-modular, 2022-07-09:
What is the proper way to install modules? I am getting this type of error:
```
Collecting micropython-ssd1306
  Using cached micropython-ssd1306-0.3.tar.gz (1.8 kB)

ERROR: micropython-ssd1306 from https://files.pythonhosted.org/packages/01/d0/0841d47772962c80af3ab178ef062ed2cd524cb99eb38463e669428402a8/micropython-ssd1306-0.3.tar.gz#sha256=d00cae1ff680e6ae504e258e700f002874b0d924b1a604253b6ea63144b58289 does not appear to be a Python project: neither 'setup.py' nor 'pyproject.toml' found.
```

gabemorris12, 2022-07-15:
I have been wondering this same thing. By the looks of it, you should use `upip` (docs found [here](https://docs.micropython.org/en/latest/reference/packages.html#packages)), but running `upip` is throwing errors for me. I posted this issue [here](https://raspberrypi.stackexchange.com/questions/138331/upip-is-not-installed), but I'm wondering if it could be an issue with the plugin.

noises-modular, 2022-07-15:
Does upip run on the microcontroller? If so, then it would likely need a network connection to connect to a repository. I have some Pi Pico W's on the way and this is one thing I plan to try.