JetBrains/intellij-micropython #201: Environment Variables?

gamename, 2022-08-09:
Hi,

Is there any way to pass environment variables to the micropython environment?  I'm trying to avoid hard-coding wifi passwords in the code.  If I could pass that as an env var, then I could just set the variable that way.

thank you

vekunz, 2022-08-09:
I am not sure how such a thing could be possible. Since a Microcontroller has no operating system, any needed data has to be saved either directly in your code or in some other connected storage.

In the case of the Raspberry Pi Pico (I am sure it works with other microcontrollers also), some hacks exist to write data directly to the flash storage, but this is highly not recommended. I am currently creating a project where I need configuration data, also (including Wifi credentials). I am going to try to use an SD card for this.

Btw: Environment variables are the least secure way to store credentials ;)

vekunz, 2022-08-09:
I think the only good way (except in source code) is an SD card.

vdavez, 2022-09-28:
I recently used this pattern: 

- create a `settings.py` or `config.py` file
- add it to .gitignore
- define your variables on each line like `SSID_NAME="Network Name"`
- copy `settings.py` on to the device
- put `import settings` at the top of `main.py`

Hope this helps!