JetBrains/intellij-micropython #204: micropython // pico // pycharm // how to upload and include classes in separate files

christianbl8m, 2022-09-14:
Hello

I am having trouble when I try to upload classes when they are in separate files from main. That is to say, the code works fine when everything is in main.py, but once I separate the classes to different files the code stops working. 
I import as I would in ordinary Python: from Metrofilter import MetroFilter
I assume that the Metrofilter.py does not work because its not there. Can I somehow include it in the Flash main.py or otherwise upload it to the pico?

soulraven, 2023-05-08:
+1 same here, is normal to say that is normal, but must be a way to import the modules autoh and keep track of them on the hardware dev.

gabemorris12, 2023-05-08:
What you need to do is put all your dependent modules (like the one that contains `MetroFilter`) into a folder. Right click on that folder and go down to "Mark directory as" and select "Sources Root". The folder icon to the left of the folder name should turn blue. Now instead of flashing `main.py`, right click on the folder in the tree and select "Flash <folder name>". This puts all the modules in the folder in the Pico (or whatever device being used). Note, you still need to have a file in the folder called `main.py`, as this is the script that will be running on the device.

soulraven, 2023-05-09:
> What you need to do is put all your dependent modules (like the one that contains `MetroFilter`) into a folder. Right click on that folder and go down to "Mark directory as" and select "Sources [...]

hello,
and how you do the import?

because relative import is not working.

my project has a lib folder that i have made as Source Root.
and is uploaded inside lib folder from esp8266 fs.

K4zuki, 2023-11-12:
Hi @soulraven ,

Is this what you expect to have?
```
<root>
|- main.py
|- Metrofilter.py
:
```

Can you try check from REPL if files are properly uploaded by `import os; os.listdir(".")`?
Following is my case:

``` python
>>> import os; os.listdir(".")
['MicroScpiDevice.py', 'RaspberryScpiPico.py']
```

if this lists what is needed, you also try `from Metrofilter import MetroFilter` from REPL and 
there should be no errors. This means the same statement in `main.py` will work.