JetBrains/intellij-micropython #205: Doesn't work with latest PyCharm (Build #PY-222.4167.33, built on September 15, 2022)

gamename, 2022-09-17:
Here is the build info:
```
PyCharm 2022.2.2 (Professional Edition)
Build #PY-222.4167.33, built on September 15, 2022
Licensed to Tennis Smith
Subscription is active until February 25, 2023.
Runtime version: 17.0.4+7-b469.53 x86_64
VM: OpenJDK 64-Bit Server VM by JetBrains s.r.o.
macOS 12.6
GC: G1 Young Generation, G1 Old Generation
Memory: 2048M
Cores: 8
Metal Rendering is ON
```

Any project that uses the micropython plugin will simply not open.

liminspace, 2022-09-21:
is the plugin dead?

gamename, 2022-09-21:
Can't tell.  The project with the plugin won't open unless the plugin is disabled first.

vekunz, 2022-09-22:
> is the plugin dead?

I think that the big problem for JetBrains currently is, that they lost many Russian developers because of obvious reasons.

octaprog7, 2022-09-22:
Debian 11. The python project does not open if the micropython plugin is enabled.

PyCharm 2022.2.2 (Community Edition)
Build #PC-222.4167.33, built on September 15, 2022
Runtime version: 17.0.4+7-b469.53 amd64
VM: OpenJDK 64-Bit Server VM by JetBrains s.r.o.

octaprog7, 2022-09-22:
I reverted back to PyCharm 2022.2.1 (Community Edition). This is currently the fastest solution to the problem. To go back to a previous version, use the JetBrains ToolBox!

Elizaveta239, 2022-09-22:
Hi! No, plugin isn't dead, we still support it. Going to release the new version soon! Thank you for your feedback!

Elizaveta239, 2022-09-23:
The issue is fixed in the plugin version `1.3.2`

closed 2022-09-23 by Elizaveta239