JetBrains/intellij-micropython #208: missing output when running code

Revilum, 2022-10-29:
when I flash a simple hello world named main.py on the raspberry pi pico I don't get any stdout.
[image]
interestingly when connecting with REPL and reset the device with CTRL+D hello world appears!
[image]

looks like the ouput cuts off after requesting a reset?!

vekunz, 2022-10-29:
This is normal. To see the output, you have to open a REPL yourself. The Flash window just shows the flashing progress and never the output.

Revilum, 2022-10-29:
ok thanks. This seems like a bit of an unnecessary hustle for debugging code.

closed 2022-10-29 by Revilum

vekunz, 2022-10-29:
Yes, currently the MicroPython support is very basic. But this may get optimized after the PR https://github.com/JetBrains/intellij-micropython/pull/139 is ready.