JetBrains/intellij-micropython #209: mac pycharm + micropython, run the program normally, but do not show the content of print('')

zskangus, 2022-11-04:
mac pycharm + micropython, run the program normally, but do not show the content of print('')

vekunz, 2022-11-04:
What are doing you in the program? How do you run the program? What microcontroller do you use? Where do you expect the print() to be shown? You have to give us more information.

zskangus, 2022-11-04:
I installed micropython plugin on mac version of pycharm to build a development environment for ESP8266. micropython firmware is also burned in ESP8266.

I can upload and run programs normally, for example, I can control the ESP's LED flashing light.

I found a problem when I was trying to do WIFI connection. I wrote in the code print('connecting to network...') , print("Connection failed!!! The wireless network has been connected for more than 15 seconds, please check if the wireless network name and password are correct...")
The messages such as "Connecting to network..." are not visible in the runtime or terminal.
But the same test items are visible when run in thonny

vekunz, 2022-11-04:
Maybe it is the same thing as here https://github.com/JetBrains/intellij-micropython/issues/208?