JetBrains/intellij-micropython #210: OSError: -6 when using mip.install in REPL, Raspberry Pico W

ElToro1966, 2022-11-08:
I have problems installing certain packages via the REPL. When I try to install, say, umqtt.simple or ulogging on my Raspberry Pico W, I get:
```
>>> import mip
>>> mip.install("umqtt.simple")
Installing umqtt.simple (latest) from https://micropython.org/pi/v2 to /lib
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "mip.py", line 166, in install
  File "mip.py", line 153, in _install_package
  File "mip.py", line 96, in _install_json
  File "urequests.py", line 180, in get
  File "urequests.py", line 76, in request
OSError: -6
```

What am I missing? MicroPython newbie...

emcek, 2022-12-31:
Did you use: [Raspberry Pi Pico W](https://micropython.org/download/rp2-pico-w/rp2-pico-w-latest.uf2) (with [urequests](https://makeblock-micropython-api.readthedocs.io/en/latest/public_library/Third-party-libraries/urequests.html) and [upip](https://docs.micropython.org/en/latest/reference/packages.html) preinstalled)

See:
https://www.raspberrypi.com/documentation/microcontrollers/micropython.html

mhixson, 2023-01-08:
Hi, I'm having this same problem. Doesn't matter what I pass to mip.install() it produces this exact error. Yes, I installed Raspberry Pi Pico W (with urequests and upip preinstalled).
Also, if I try loading https://micropython.org/pi/v2 in a web browser I get a 403 Forbidden. Is that normal?

ElToro1966, 2023-01-24:
> Did you use: [Raspberry Pi Pico W](https://micropython.org/download/rp2-pico-w/rp2-pico-w-latest.uf2) (with [...]

Also used Raspberry Pi Pico W with urequests and upip preinstalled

dfluegge, 2023-02-03:
I got it to work by first manually establishing a wifi connection between Pico PI and home LAN using REPL. Once it is connected, mip.install worked fine. 

import network
wlan = network.WLAN(network.STA_IF)
wlan.active(True)
wlan.connect(ssid, password)

import mip
mip.install("umqtt.simple")

elmot, 2024-08-29:
This is not the plugin issue

closed 2024-08-29 by elmot