JetBrains/intellij-micropython #214: REPL Start

John1900, 2022-12-09:
After making a code change and starting the run (shift F10) is there any way to automatically starting the Repl display of print statements?

The process:  menu/tools/micropython/click in repl window/ctl-D restarts the code

Since my program makes things move (stepper, servo) there is motion BEFORE cntl-D which makes debugging near impossible.

Thanks

aka13-404, 2023-03-27:
What I would like, is at least for something to tell me, that I have to do a Ctrl-D to rerun the program! 
I have spent a day trying to figure out, why my print() instructions are worse than useless :D 
Upload a program, go to repl - nothing happens, restart repl session - program is "dead", interactive prompt only. 
I even went as far as to switch IDEs, until I found this ticket with the Ctrl+D suggestion. 
 
Obviously, having to reset manually is a pain, and I would love for it to automatically open and start repl after code upload.

John1900, 2023-03-29:
Gave up because of the problems debugging - see my first post. Now use Thonny which works but not nearly as convenient as pycharm.

elmot, 2024-07-24:
Duplicate #283

elmot, 2024-07-27:
Now fixed in release [1.4.5](https://github.com/JetBrains/intellij-micropython/releases/tag/1.4.5)