JetBrains/intellij-micropython #216: REPL does not respond to cursor keys and CTRL+L for some combination of OS and IDE

K4zuki, 2022-12-11:
### REPL does not respond to cursor keys and CTRL+L for some combination of OS and IDE. 

Some are ok, some are NG. There is no major configuration difference I can tell.
It responds to Tab and Backspace keys.

What am I missing?

#### Environments:
4x combinations. 1x Mac, 3x Windows; 1x2022.3, 3x2022.1, 

#### BAD cases
1. Windows + 2022.3 professional
```
PyCharm 2022.3 (Professional Edition)
Build #PY-223.7571.203, built on November 30, 2022
Licensed to *** / ***
Subscription is active until ***
Runtime version: 17.0.5+1-b653.14 amd64
VM: OpenJDK 64-Bit Server VM by JetBrains s.r.o.
Windows 10 10.0
GC: G1 Young Generation, G1 Old Generation
Memory: 2040M
Cores: 8
Non-Bundled Plugins:
    com.dubreuia (2.3.0)
    cn.haojiyou.CodeGlance3 (2.1.1-2021.1.3)
    SerialPortMonitor (223.7571.203)
    com.sburlyaev.terminal.plugin (0.4.5)
    name.kropp.intellij.makefile (223.7571.203)
    color.scheme.Visual Studio Dark (1.0.0)
    com.intellij.plugin.adernov.powershell (2.0.10)
    YAML/Ansible support (0.11.2)
    com.samdark.intellij-visual-studio-code-dark-plus (2.6)
    PlantUML integration (5.20)
    color.scheme.VisualStudio2019Dark (1.0.0)
    net.seesharpsoft.intellij.plugins.csv (3.0.1-223)
    intellij-micropython (1.4.0)
    mobi.hsz.idea.gitignore (4.4.4)
```
2. Windows + 2022.1 community
```
PyCharm 2022.1 (Community Edition)
Build #PC-221.5080.212, built on April 12, 2022
Runtime version: 11.0.14.1+1-b2043.25 amd64
VM: OpenJDK 64-Bit Server VM by JetBrains s.r.o.
Windows 10 10.0
GC: G1 Young Generation, G1 Old Generation
Memory: 2032M
Cores: 8
Non-Bundled Plugins:
    net.seesharpsoft.intellij.plugins.csv (2.21.0)
    name.kropp.intellij.makefile (221.5080.126)
    com.tang (1.3.6.251-IDEA213)
    com.sburlyaev.terminal.plugin (0.4.5)
    com.samdark.intellij-visual-studio-code-dark-plus (2.6)
    com.intellij.plugin.adernov.powershell (2.0.10)
    com.dubreuia (2.3.0)
    color.scheme.Visual Studio Dark (1.0.0)
    cn.haojiyou.CodeGlance3 (2.1.1-2021.1.3)
    SerialPortMonitor (0.7)
    izhangzhihao.rainbow.brackets (6.26)
    intellij-micropython (1.4.0)
    PlantUML integration (5.19)
    mobi.hsz.idea.gitignore (4.4.0)
    Docker (221.5080.126)
    Batch Scripts Support (1.0.13)
```

#### GOOD cases
1. Windows + 2022.1 community
```
PyCharm 2022.1 (Community Edition)
Build #PC-221.5080.212, built on April 12, 2022
Runtime version: 11.0.14.1+1-b2043.25 amd64
VM: OpenJDK 64-Bit Server VM by JetBrains s.r.o.
Windows 10 10.0
GC: G1 Young Generation, G1 Old Generation
Memory: 2022M
Cores: 12
Registry:
    ide.balloon.shadow.size=0

Non-Bundled Plugins:
    net.seesharpsoft.intellij.plugins.csv (2.21.0)
    name.kropp.intellij.makefile (221.5080.126)
    com.sburlyaev.terminal.plugin (0.4.5)
    com.samdark.intellij-visual-studio-code-dark-plus (2.6)
    com.pandocker (0.7.3)
    com.intellij.plugin.adernov.powershell (2.0.10)
    com.intellij.jira (4.0.1)
    com.dubreuia (2.3.0)
    color.scheme.Visual Studio Dark (1.0.0)
    cn.haojiyou.CodeGlance3 (2.1.1-2021.1.3)
    SerialPortMonitor (0.7)
    izhangzhihao.rainbow.brackets (6.26)
    intellij-micropython (1.4.0)
    PlantUML integration (5.19)
    mobi.hsz.idea.gitignore (4.4.0)
    git-commit-message-plugin (1.3)
    Docker (221.5080.126)
```
2. Mac + 2022.1 professional
```
PyCharm 2022.1 (Professional Edition)
Build #PY-221.5080.212, built on April 12, 2022
Licensed to *** / ***
Subscription is active until ***
Runtime version: 11.0.14.1+1-b2043.25 x86_64
VM: OpenJDK 64-Bit Server VM by JetBrains s.r.o.
macOS 11.4
GC: G1 Young Generation, G1 Old Generation
Memory: 2048M
Cores: 8
Non-Bundled Plugins:
    [http://org.turbanov.run.configuration.as](https://t.co/FwhzkDoMgB).action (1.4.5)
    name.kropp.intellij.makefile (221.5080.126)
    monokai-pro (1.9)
    [http://com.xinkun.theme.md](https://t.co/IHe9rkMD14) (1.7.12)
    com.tang (1.3.6.251-IDEA213)
    com.samdark.intellij-visual-studio-code-dark-plus (2.6)
    com.pandocker (0.7.3)
    com.intellij.javafx (1.0.4)
    com.github.b3er.idea.plugins.arc.browser (0.32)
    com.dubreuia (2.3.0)
    cn.haojiyou.CodeGlance3 (2.1.1-2021.1.3)
    YAML/Ansible support (0.11.2)
    SerialPortMonitor (0.7)
    PlantUML integration (5.19)
    mobi.hsz.idea.gitignore (4.4.0)
    NodeJS (221.5080.169)
    izhangzhihao.rainbow.brackets (2022.3.4-ij)
    intellij-micropython (1.4.0)
```

jetsen-os, 2022-12-21:
Same here...

It is strange that some keys work via certain key combinations. For example: CTRL + B corresponds to "left key".

See my config:
[config.txt](https://github.com/JetBrains/intellij-micropython/files/10276542/config.txt)

Wind-stormger, 2023-01-13:
So am I. 
I'm confused about not being able to use the cursor keys.

> It is strange that some keys work via certain key combinations. For example: CTRL + B corresponds to "left key".

Ctrl+p seems to be the up key.

jetsen-os, 2023-06-20:
I have a solution for myself...

In my opinion, there is an error in `serial.tools.miniterm.py` (pyserial). Here is the description of the `msvcrt.getwch()` method ( https://docs.python.org/3/library/msvcrt.html ):

> msvcrt.getch()
> Read a keypress and return the resulting character as a byte string. Nothing is echoed to the console. This call will block if a keypress is not already available, but will not wait for Enter to be pressed. If the pressed key was a special function key, this will return '\000' or '\xe0'; the next call will return the keycode. The Control-C keypress cannot be read with this function.
> 
> msvcrt.getwch()
> Wide char variant of [getch()](https://docs.python.org/3/library/msvcrt.html#msvcrt.getch), returning a Unicode value.

"F1", "F2" etc. as well as the "up" key etc. are special function keys. Why does `miniterm.py` distinguish between them (`fncodes`, `navcodes`)??

I changed `return self.fncodes[code]` to `return self.navcodes[code]` and now everything works.

I hope it will help to fix the bug.

bc-109, 2023-09-18:
Hi all,

I have this problem with current versions of PyCharm under Windows, and that's quite annoying.  Up / down arrow keys do not work for command history in REPL. When using an external terminal (Putty, SecureCRT), it works. 

Are there currently any workaround for that ?

The most obvious one would be to use external terminal for REPL. But as far as serial port is used by Pucharm "Run" to upload / flash software to the board, that would not work if serial port is in use by an external terminal. "Run/debug configuration" allows ro run external command before launch (to kill external terminal). But after launch, the only option is "open REPL on success". Maybe it would be a great idea to be able to launch an external command, so that we can run an external terminal instead.

Thank you in advance. Kind regards

zerovijay, 2024-01-01:
Micropython REPL not working on new version pycharm. This is temporary issue
Maybe it's well fixed on next update

bc-109, 2024-02-14:
REPL now works again with latest PyCharm 2023.3.3.
Anyway, cursor keys still do not work on my machine (Windows 10). Exactly as before.