JetBrains/intellij-micropython #217: Missing "micropython" under "Languages & Frameworks"

ziGuy, 2022-12-18:
Hi,

I've just got my first esp32 device (ESP32-C3), and trying to use micropython plugin without any success.
Installed the plugin successfully, but in 
In Settings > Editor > Inspections > MicroPython, i see the following warning:
"**Detects missing Python packages that are required for communicating with"**

Under Settings > Languages & Frameworks, i should see "micropython" but i don't.

When trying to connect by esptool (https://github.com/espressif/esptool.git), it seems to succeed:

```
Serial port /dev/ttyACM0
Connecting...
Detecting chip type... ESP32-C3
Chip is ESP32-C3 (revision v0.3)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 60:55:f9:c2:b9:50
Uploading stub...
Running stub...
Stub running...
Status value: 0x0200
Hard resetting via RTS pin...
```

Any help will be appreciated
I'm using: Linux debian 4.9.0-8-amd64 #1 SMP Debian 4.9.130-2 (2018-10-27) x86_64 GNU/Linux

jcjveraa, 2023-01-18:
I have exactly the same behavior unfortunately - no good indication as to what might be going wrong, it's 'just not there'

flaviut, 2023-01-20:
Would you please take a look @Elizaveta239? Some more background:

Nothing interesting in idea.log with `#com.jetbrains.micropython` set in `Help | Diagnostic Tools | Debug Log Settings`.

```
IntelliJ IDEA 2022.3.1 (Ultimate Edition)
Build #IU-223.8214.52, built on December 20, 2022
Licensed to ...
Subscription is active until May 10, 2023.
Runtime version: 17.0.5+1-b653.23 amd64
VM: OpenJDK 64-Bit Server VM by JetBrains s.r.o.
Linux 6.1.6-arch1-3
GC: G1 Young Generation, G1 Old Generation
Memory: 2048M
Cores: 16
Registry:
    ide.browser.jcef.enabled=false
    debugger.watches.in.variables=false
    debugger.new.tool.window.layout=true
    ide.experimental.ui=true

Non-Bundled Plugins:
    com.jetbrains.plugins.ini4idea (223.8214.64)
    org.rust.lang (0.4.186.5143-223)
    org.intellij.plugins.hcl (223.8214.6)
    com.intellij.plugins.watcher (223.7571.113)
    com.github.copilot (1.1.38.2229)
    Key Promoter X (2022.3)
    org.intellij.scala (2022.3.16)
    Pythonid (223.8214.52)
    izhangzhihao.rainbow.brackets (2023.1.1)
    com.google.idea.bazel.ijwb (2022.12.13.0.1-api-version-223)
    intellij-micropython (1.4.0)
    io.snyk.snyk-intellij-plugin (2.4.50)
    uk.co.ben-gibson.remote.repository.mapper (4.3.1)

Kotlin: 223-1.7.21-release-272-IJ8214.52
Current Desktop: Undefined
```

Using PyCharm works around this problem.

Elizaveta239, 2023-03-17:
Hi! Sorry for the late response! So the issue is that the plugin doesn't work with Intellij Idea 2022.3, but works fine with PyCharm 2022.3, right?

flaviut, 2023-03-17:
Yes, that's right.

Elizaveta239, 2023-03-17:
Hi! I've got the problem. In PyCharm the plugin works on project level, but in Intellij Idea it works on module level. So you need to configure device-related settings separately, and you can do it in `Project Structure -> Project settings -> Modules -> <select your module> -> MicroPython`

[image]

Feel free to let me know if you have any questions!

closed 2023-03-17 by Elizaveta239