JetBrains/intellij-micropython #223: Incorrect upload path for flashed file, RPI Pico, PyCharm 2022.1.4

philip225, 2023-01-08:
When attempting to flash over a file, the path does not start from the project or marked source route directory, it instead includes the whole project path, from the OS root.
You can observe this in the file upload status messages, the path shown for an example project:
/home/philip/projects/python/example_micro/main.py -> ../../../../../home/philip/projects/python/example_micro/main.py

This results in the project ending up in /home/philip/projects/python/example_micro/ on the Pico, instead of /
Notice also the strange 5 prepended '../', which matches the path elements counting from /home/

PyCharm 2022.1.4 community edition on Linux.
RPI Pico running Micropython v1.19.1-782-g699477d12

philip225, 2023-01-08:
More information:
The root issue is with the call to microupload.py, the wrong path for the micropython device, is being passed to microupload.py.
The final parameter should be the file being uploaded and should have a relative path, not an absolute path.
If I run microupload.py manually with the corrected path, it works as expected.
I suspect the '../' is a symptom of the issue.
Doing some further digging it looks like whatever is passed for the 'path' string type parameter in getMicroUploadCommand in MicroUpload.kt is wrong, why I don't know.

nyirit, 2023-10-22:
This issue is still happening. I arrived pretty much at the same conclusion as Philip.

- OS: Linux Mint
- PyCharm 2023.2.3 (CE)
- MicroPython plugin: 1.4.2-2023.2
- Board: Wemos D1 mini
- MicroPython version: ESP8266_GENERIC-OTA-20231005-v1.21.0

elmot, 2024-11-09:
Please try NOVA version: #330