JetBrains/intellij-micropython #228: ESP8266 D1 mini reboot loop after flash

Saunakalja, 2023-02-17:
I have simple blink program that is working.
```
from machine import Pin
from time import sleep
import sys
led = Pin(2, Pin.OUT)
counter = 0
print("Lets blink!")
while True:
    print("Count:", counter)
    counter += 1
    led(1)
    sleep(1)
    led(0)
    sleep(1)
```
I can flash it without problems.
```
Connecting to /dev/ttyUSB0
Uploading files: 0% (0/1)
/media/storage/python/micropythonProject1/main.py -> main.py
Uploading files: 100% (1/1)
Soft reboot
```
After flash REPL shows long list of:
```
Count: 0
Lets blink!
Count: 0
Lets blink!
Count: 0
Lets blink!
Count: 0
Lets blink!
Count: 0
Lets blink!
Count: 0
```
Then multiple lines of this:
```
Traceback (most recent call lastTraceback (most recent call lastTraceback (most recent call lastTraceback (most recent call lastTraceback (most recent call lastTraceback (most recent call lastTraceback (most recent call lastTraceback (most recent call lastTraceback (most recent call lastTraceback (most recent call lastTraceback (most recent call lastTraceback (most recent call lastTraceback (most recent call lastTraceback (most recent call lastTraceback (most recent call lastTraceback (most recent call lastTraceback (most recent call lastTraceback (most recent call lastTraceback (most recent call lastTraceback (most recent call lastTraceback (most recent call lastTraceback (most recent
```
Then:
```
cent call lastTraceback (most recent call lastTraceback (most recent call lastTraceback (most recent call lastTraceback (most recent call lastTraceback (most recent call lastTraceback (most recent call lastTraceback (most recent call lastTrace File "main.py", line 11, in <module>
KeyboardInterrupt: 
MicroPython v1.19.1 on 2022-06-18; ESP module with ESP8266
Type "help()" for more information.
>>> 

```
If I now hit Ctrl+D it soft reboots and starts working like it should.

```
>>> 
MPY: soft reboot
Lets blink!
Count: 0
Count: 1
Count: 2
Count: 3
Count: 4
Count: 5
```
Any trick to get rid of that extra Ctrl-D after flash? It does that soft reboot automatic after flash but still I have to do it manually. With Thonny I don't have this problem.

Micha10, 2023-03-24:
I have the same issue. 
flashing sources. 
```
Device path COM3
Quit: Ctrl+] | Stop program: Ctrl+C | Reset: Ctrl+D 
Type 'help()' (without the quotes) then press ENTER.
���p�N�|r��n|�$
               ld$`"�
Starte Main    
Ende Main                            
Traceback (most recent call last):   
  File "main.py", line 9, in <module>
KeyboardInterrupt:                   
                                     
MicroPython v1.19.1 on 2022-06-18; ESP module with ESP8266
Type "help()" for more information.                       
>>>                                      
```

So everytime after flashing it crashes. 

If I now use CTRL-D a soft reboot is made and the programm runs correctly

13849131939, 2023-04-07:
I have the same issue.
PyCharm 2023.1 (Community Edition)
Build #PC-231.8109.197, built on March 29, 2023

lp20010415, 2023-05-17:
> I have the same issue. flashing sources:
>
> ```
> Device path COM3
> Quit: Ctrl+] | Stop program: Ctrl+C | Reset: Ctrl+D
> Type 'help()' (without the quotes) then press ENTER. [...]

Yeah, So do I. Using CTRL+D works, so I thinking about how to make the program automatical finish it.