JetBrains/intellij-micropython #232: [Help] IntelliJ Idea setup

Alemiz112, 2023-04-16:
Hi, 
I can not figure out how to setup Intellij Idea with Micropython. I have got both Python and Micropython plugins installed. But I got stuck at `Add the MicroPython facet to a Python module in your project structure and specify the path to your MicroPython device` step.

When I go to project settings -> facets and try to add new facet, error window with `No suitable modules for Python facet found.` appears.
[image]

Is anyone experienced with setting Micropython in IntelliJ able to help?
Thank you!

Elizaveta239, 2023-04-19:
Hi! In Intellij Idea you need to configure device-related settings separately, and you can do it in `Project Structure -> Project settings -> Modules -> <select your module> -> MicroPython`

[image]

After that the device path will be configured and facet will be configured automatically

Alemiz112, 2023-04-19:
Hi, thank you for reply. The problem is that I do not see MicroPython under my modules tab.

dawiemalan, 2023-04-21:
Same here. MicroPython facet is not available.

IntelliJ 2023.1 Ultimate on Ubuntu.

taxpon, 2024-06-02:
Same here. I don't see MicroPython in the facets section. Mine is IntelliJ IDEA 2024.1.