JetBrains/intellij-micropython #237: RP2: PIO commands in a python file are flagged as "Unresolved reference" errors

sandyscott, 2023-05-31:
If you have a [PIO](https://docs.micropython.org/en/latest/rp2/tutorial/pio.html) program assembled in a python file, all of the instuctions are flagged as "Unresolved reference" errors.
```python
import rp2

@rp2.asm_pio()
def quadrature_encoder():
    jmp("label1")
    nop()
    label("label1")
```
appears as
[image]

This issue applies to all instructions and directives appearing the program.

The full list of instructions and directives is visible in the official documentation:
[Programmable IO - MicroPython latest documentation](https://docs.micropython.org/en/latest/rp2/tutorial/pio.html)

fiechr, 2023-06-04:
I'm having the same issue.

Example:

```python
from machine import Pin
from rp2 import PIO, StateMachine, asm_pio
from time import sleep

@asm_pio(sideset_init=PIO.OUT_LOW)
def pwm_prog():
    pull(noblock) .side(0)
    mov(x, osr) # Keep most recent pull data stashed in X, for recycling by noblock
    mov(y, isr) # ISR must be preloaded with PWM count max
    label("pwmloop")
    jmp(x_not_y, "skip")
    nop()         .side(1)
    label("skip")
    jmp(y_dec, "pwmloop")
```

[image]

There are other error hints as well. It seems to me, that maybe the bundled µPython isn't the latest version? Can this be changed by the user? For example the class StateMachine doesn't have a exec() method, although in newer version it seems to have it, since the code runs completely fine when just flashing it to the RP2040.

gabemorris12, 2025-01-19:
PIO programs are a very special case where the contents do not comply with Python syntax. I'm not sure if this plugin will ever support this kind of recognition, but in the meantime, just add the `# noinspection All` comment to the top of the function. This will at least make it bearable.

[image]

Josverl, 2025-01-20:
Actually in the form above the PIO code is in python syntax, and can be parsed with the correct type stubs.

See: https://github.com/Josverl/PIO_ASM_typing

I will be adding that to the type-stubs for micropython v1.24, in the next weeks