JetBrains/intellij-micropython #243: REPL console not working after update to PyCharm 2023.2CE

doccodyblue, 2023-08-01:
After updating to build PC-232.8660.197 (running on Mac Sonoma) the REPL console doesn't connect anymore.

Error message is "Device path is not specified, please check settings." but the settings should be ok.
Uploading to the miropython device still works as expected. 

For now I use the Serial Monitor which is working - so the serial connection is not an issue.

gabemorris12, 2023-08-01:
There is an issue with the REPL not working at all for the latest update (#242 ). Until that gets fixed, you can download the jetbrains toolbox app and install the previous version, or you can install the previous version directly from their site.

lancelote, 2023-10-16:
REPL functionality is fixed in #253.

closed 2023-10-16 by lancelote